NixOS/nixos-hardware #340: Dell XPS 9310 bluetooth issue on Linux 5.15.3

hissssst, 2021-11-24:
```nix
  boot.kernelPackages = lib.mkIf (lib.versionOlder pkgs.linux.version "5.12") (lib.mkDefault pkgs.linuxPackages_latest);
  boot.kernelPatches = [{
    name = "enable-qca6390-bluetooth";
    patch = null;
    extraConfig = ''
      BT_QCA m
      BT_HCIUART m
      BT_HCIUART_QCA y
      BT_HCIUART_SERDEV y
      SERIAL_DEV_BUS y
      SERIAL_DEV_CTRL_TTYPORT y
    '';
  }];
```

Updating to Linux 5.15.3 loses bluetooth adapter. `bluetoothctl` shows `no default controller available`
Rollback to `5.15.0` fixes the problem

I can debug and provide additional info, just tell me what to do

@mitchmindtree

mitchmindtree, 2021-11-25:
@hissssst hmmm I just updated to 5.15.4 last night and all seems to be working well. Perhaps it was just that patch release?

Could you share which nix-channel you are on? E.g. the output of `sudo nix-channel --list`? I wonder if maybe you've picked up NixOS/nixpkgs#144409 and nixos doesn't like that the config is both in nixpkgs and nixos-hardware... though I would imagine these would all get merged anyway. If updating to 5.15.4 doesn't work, you could potentially try using #341 and changing your nixos channel to that branch as I mention in the PR?

hissssst, 2021-11-25:
I am on `nixos https://nixos.org/channels/nixos-unstable`
Let me try `5.15.4`

hissssst, 2021-11-25:
@mitchmindtree , yeah `5.15.4` works as intended

closed 2021-11-25 by hissssst