NixOS/nixos-hardware #348: Nvidida power management configuration

grnnja, 2021-12-17:
Looking through the repository, none of the configurations have `hardware.nvidia.powerManagement` configured. On my dell machine,  I've been able several watts with;

```
  hardware.nvidia = {
    powerManagement = { 
      enable = true;
      finegrained = true;
    };
    nvidiaPersistenced = true;
  };
```
These options control these things from the arch wiki: https://wiki.archlinux.org/title/PRIME#PCI-Express_Runtime_D3_(RTD3)_Power_Management 

I'm wondering if not including this configuration is intentional or not? I'm not sure if this counts as an opinionated setting. 
If it is okay I would like to submit a pr for my machine.

Mic92, 2021-12-17:
I don't think this is intentional, at least I am not aware of that. Could be added here? https://github.com/NixOS/nixos-hardware/blob/2a7063461c3751d83869a2a0a8ebc59e34bec5b2/common/gpu/nvidia.nix#L19

grnnja, 2021-12-18:
From reading the nvidia docs it seems there is less to configure that I thought. The configuration is dependent on the microarchitecture. 
For cards earlier than turing (below 20 series) there is no power saving and the configuration should be:
```
  hardware.nvidia = {
    powerManagement = { 
      enable = true;
    };
  };
 ```
 For turing (20 series) cards it should be:
 ```
   hardware.nvidia = {
    powerManagement = { 
      enable = true;
      finegrained = true;
    };
    nvidiaPersistenced = true;
  };
  ```
and for ampere (30 series) and later cards power saving is enabled by default so it is just:
```
  hardware.nvidia = {
    powerManagement = { 
      enable = true;
    };
  };
 ```
I'm not 100% sure on this and I only have my laptop to test but I think they are right.
 So maybe its best set `  hardware.nvidia.powerManagement.enable = true;` is in
  https://github.com/NixOS/nixos-hardware/blob/2a7063461c3751d83869a2a0a8ebc59e34bec5b2/common/gpu/nvidia.nix#L19
 and for laptops with turing cards we can enable the other things in their configuration?

Mic92, 2021-12-20:
There could be a subprofile for turing cards that is more specific.

grnnja, 2021-12-20:
Yeah I do.

grnnja, 2021-12-20:
Should I submit a pr with another file in the gpu folder with the turing configuration and add the enable the powerManagement option in the nvidia file?