NixOS/nixos-hardware #354: Issues building configuration for Thinkpad T430 (boot.kernelParams)

eunjibyte, 2022-01-04:
Line in question:

https://github.com/NixOS/nixos-hardware/blob/3aabf78bfcae62f5f99474f2ebbbe418f1c6e54f/lenovo/thinkpad/t430/default.nix#L13

Error provided when building configuration:

```
·> sudo nixos-rebuild switch
building the system configuration...
error: A definition for option `boot.kernelParams.[definition 1-entry 1]' is not of type `string, with spaces inside double quotes'. Definition values:
- In `/nix/var/nix/profiles/per-user/root/channels/nixos-hardware/lenovo/thinkpad/t430': "acpi_osi='!Windows 2012'"
```

This appears similar to https://github.com/NixOS/nixos-hardware/issues/326 (NixOS 21.11) but without flakes.

turboMaCk, 2022-01-05:
seems like a breaking change upstream. But I completely missed that we now have multiple types of string.

rnhmjoj, 2022-01-05:
Here's the change: https://github.com/NixOS/nixpkgs/pull/133625

WolfangAukang, 2022-01-11:
@eunjibyte the line you are pointing seems to be prior to the fix applied to #326, or at least if I click on the line you indicate, [this is the latest commit](https://github.com/NixOS/nixos-hardware/commit/a8ea967866515cee5a4281ac7799bd82219e5fb7). [This is the content on master right now](https://github.com/NixOS/nixos-hardware/blob/master/lenovo/thinkpad/t430/default.nix).

How are you trying to import these settings?

eunjibyte, 2022-01-12:
@WolfangAukang Thanks for noting that the fix is live.

I somehow removed the `nixos-hardware` channel. Solved this issue by re-adding and upgrading it before rebuilding.

closed 2022-01-12 by eunjibyte