NixOS/nixos-hardware #362: Surface Pro 8 support

kiana-S, 2022-01-18:
The `microsoft/surface` module currently does not provide support for the keyboard and touchpad of the Microsoft Surface Pro 8, despite the fact that the [linux-surface feature matrix](https://github.com/linux-surface/linux-surface/wiki/Supported-Devices-and-Features#feature-matrix) states that it should.

I'm not entirely sure why this is, but I believe that it is because that module uses the Linux 5.13 kernel, and the patches for the Surface Pro 8 are [only present on release 5.14](https://github.com/linux-surface/linux-surface/issues/616#issuecomment-955812230) (I could be wrong on that, though).

Updating the Linux kernel and patches to a newer version may solve this issue.

Mic92, 2022-01-25:
Patches are welcome if you update the linux kernel linked here.

kiana-S, 2022-03-04:
Fixed by #368 (merged into #385)

closed 2022-03-04 by kiana-S