NixOS/nixos-hardware #364: WiFi adapter not found / loading on Surface Book 2 with ZFS root

shadowrylander, 2022-01-24:
Hello!

As in the title; I can access WiFi from the installer, but not the install itself. I'm trying to install 21.11.

Thank you kindly for the help!

Mic92, 2022-01-24:
Does this help?

```
{
  hardware.enableRedistributableFirmware = lib.mkDefault true;
}
```

Also check in the output of the `dmesg` command if your wifi driver is missing firmware.

Mic92, 2022-01-24:
cc @mexisme
Also what is your wifi adapter?
```
$ nix-shell -p inxi --command 'inxi -N'
Network:   Device-1: Intel Comet Lake PCH-LP CNVi WiFi driver: iwlwifi
           Device-2: Intel Ethernet I219-V driver: e1000e
```

shadowrylander, 2022-01-24:
Checked the wifi adapter:
```
Network:   Device-1: Marvell 88W8897 [AVASTAR] 802.11ac Wireless driver: mwifiex_pcie
           Device-2: Marvell Bluetooth and Wireless LAN Composite type: USB driver: btusb
```

shadowrylander, 2022-01-24:
Oops.

Currently checking the hardware option.

shadowrylander, 2022-01-24:
Hah! It worked! Thank you so much! Should one of us submit a PR for this?

shadowrylander, 2022-01-24:
Also, could you do it, if possible? I'm off to sleep now! 😅

Mic92, 2022-01-24:
Preferable you would do it, since I cannot test it and we would rely on you to test future changes to the profile (right now we often use git blame to find testers)

shadowrylander, 2022-01-24:
Although, slight issue: you need to manually go in and enter the wifi password with network manager, directly from the settings' security tab of the wifi network. Minor complaint, though, there.

shadowrylander, 2022-01-24:
Ah; makes sense. I'll submit a PR in the morning, then! Thanks again!

closed 2022-01-24 by shadowrylander

mexisme, 2022-01-24:
> Preferable you would do it, since I cannot test it and we would rely on you to test future changes to the profile (right now we often use git blame to find testers)

I'm afraid I can't test this, either, but happy to review.

shadowrylander, 2022-01-24:
And done, @mexisme and @Mic92 !