NixOS/nixos-hardware #378: Add imx290 support to Raspberry Pi

amaumene, 2022-02-16:
Hi,

Would it be possible to add this dts for Raspberry Pi? https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/overlays/imx290-overlay.dts 

This is to support imx290 based camera.

I've tried on my side but unfortunately I don't have the skills to hack dts :(

Thanks in advance :)

Mic92, 2022-02-16:
You can probably deal with in the same we way did for other overlays: https://github.com/NixOS/nixos-hardware/blob/master/raspberry-pi/4/dwc2.nix
I don't think someone will pick up this issue unless they own the hardware realistically. So you might be better of asking questions here on matrix in the ARM channel of NixOS.

jpds, 2023-10-03:
I implemented this for the IMX477 over at https://github.com/NixOS/nixos-hardware/compare/master...jpds:nixos-hardware:rpi4-imx477 . It was indeed a case of dumping the `.dtbo` file as described in the comment and tweaking the `compatible` field.

doronbehar, 2026-08-16:
Merged PRs might help:

- #1894
- #1947