NixOS/nixos-hardware #381: Touchpad support for system76 Gazelle

tartavull, 2022-02-20:
Hello everybody,

I'm new to NixOS and I can't figure out how to enable the touchpad on my laptop.

```
services.xserver = {
    enable = true;
    autorun = true;
    exportConfiguration = true;
    #videoDrivers = ["nvidia"];
    layout = "us";
    displayManager.lightdm.enable = true;
    windowManager.i3.enable = true;
    libinput = {
      enable = true;
      touchpad.naturalScrolling = true;
      touchpad.tapping = true;
      touchpad.middleEmulation = true;
    };
    displayManager = {
      defaultSession = "none+i3";
      gdm.enable = false;
      sddm.enable = false;
    };
  };
```
adding touchpad to the configuration.nix didn't solve the issue. Any advice?

Thank you!

Mic92, 2022-06-27:
cc @khumba

closed 2022-09-05 by tartavull

zarelit, 2022-09-06:
@tartavull how did you solve the issue?