NixOS/nixos-hardware #409: `boot.kernelPatches' not respected

shadowrylander, 2022-04-21:
Hello!

As in the title; I'm trying to apply a patch for `bcachefs`, but `boot.kernelPatches` is not being respected; more information about my config can be found in [this issue](https://github.com/NixOS/nixpkgs/issues/168987#issuecomment-1101104812) over on the `nixos/nixpkgs` repo, where another user narrowed it down for me.

Thank you kindly for the help!

Mic92, 2022-06-27:
Potentially the second `callPackage` level prevents nixos from applying patches to the kernel: https://github.com/NixOS/nixos-hardware/blob/master/microsoft/surface/kernel/linux-5.16.11.nix
This is what happens in nixpkgs: https://github.com/NixOS/nixpkgs/blob/master/pkgs/os-specific/linux/kernel/linux-5.4.nix

shadowrylander, 2022-06-27:
Ah; make sense. I did the following steps to fix the issue in my fork:
* https://github.com/shadowrylander/nixos-hardware/blob/master/microsoft/surface/kernel/default.nix#L4
* https://github.com/shadowrylander/nixos-hardware/blob/master/microsoft/surface/kernel/linux-5.16.11.nix#L1
* https://github.com/shadowrylander/nixos-hardware/blob/master/microsoft/surface/kernel/linux-5.16.11.nix#L15

They seem to work for now; is there anything in that setup I should be wary of?

Mic92, 2022-07-21:
I think we could take your nixos-hardware changes as they are.

shadowrylander, 2022-07-21:
As in, should I submit a pull request?

Mic92, 2022-07-22:
Yes. https://github.com/shadowrylander/nixos-hardware/blob/667f3bccdc827e7635b474a6d9650a4f2fe6c1bb/microsoft/surface/kernel/default.nix#L4 looks like something nice to have.

shadowrylander, 2022-07-22:
Got it; I'll submit one as soon as possible!