NixOS/nixos-hardware #410: How to check if I'm running the surface kernel?

shadowrylander, 2022-04-29:
Hello!

As in the title; the surface kernel version in the build attribute set merely states the Linux kernel version, without something like a `-surface` suffix, such as `5.16.11-surface`. How can I check if I'm actually using the surface kernel?

Thank you kindly for the help!

Mic92, 2022-06-27:
Sorry for the late response. This might work:

```
ls -la /run/current-system/kernel
```

And if not than this maybe:

```
nix show-derivation /run/current-system/kernel | grep src
```

shadowrylander, 2022-06-27:
Sorry; I meant from within my `NixOS` config itself!

Mic92, 2022-08-01:
You can load up your nixos configuration in a nixos repl: https://nixos.wiki/wiki/Nix_command/repl#Examples

closed 2022-08-01 by Mic92

shadowrylander, 2022-08-01:
Ah, yes; that'll have to do. Thanks for all the help!