NixOS/nixos-hardware #434: Raspberry Pi image fails with 4K monitor + Pi 4B

adamlwgriffiths, 2022-07-06:
I've been struggling to get a Nix image on a raspberry pi.
Whenever I flash the image and boot it up, just before it gets to an interactive prompt, the screen goes black (no signal).

I've just determined this is because I'm using a 4K TV (HDMI).
When I swap to a smaller ~1920x1080 monitor (also HDMI), the system makes it past this point and presents me with an interactive prompt.

I've successfully booted this up with the latest Raspbian image and had a 4K terminal, so the Pi certainly can handle it (I know it was at 4K because it was neigh unreadable with how small the text was).
I've also updated the EEPROM firmware and bootloader (using raspbian) to no effect.

My assumption is that the Nix image cannot handle the transition during boot to 4K once it brings up the graphics card?

Hardware:
Raspberry Pi 4 Model B 2GB.
Video over HDMI.

[Discourse link](https://discourse.nixos.org/t/nix-raspberry-pi-4-images-dont-work-on-4k-monitor/20140/3)

nixos-discourse, 2022-07-06:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/nix-raspberry-pi-4-images-dont-work-on-4k-monitor/20140/4

adamlwgriffiths, 2022-07-07:
Full `dmesg` and `journalctl -b` logs [here](https://gist.github.com/adamlwgriffiths/d96e484047822653855cb1aa88a95c48)

Of note is the NULL pointer dereference that occurs when plugging in the TV.

```
Jul 07 02:32:43 nixos kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000348
Jul 07 02:32:43 nixos kernel: Mem abort info:
Jul 07 02:32:43 nixos kernel:   ESR = 0x96000006
Jul 07 02:32:43 nixos kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Jul 07 02:32:43 nixos kernel:   SET = 0, FnV = 0
Jul 07 02:32:43 nixos kernel:   EA = 0, S1PTW = 0
Jul 07 02:32:43 nixos kernel:   FSC = 0x06: level 2 translation fault
Jul 07 02:32:43 nixos kernel: Data abort info:
Jul 07 02:32:43 nixos kernel:   ISV = 0, ISS = 0x00000006
Jul 07 02:32:43 nixos kernel:   CM = 0, WnR = 0
Jul 07 02:32:43 nixos kernel: user pgtable: 4k pages, 48-bit VAs, pgdp=000000004a520000
Jul 07 02:32:43 nixos kernel: [0000000000000348] pgd=080000004a51c003, p4d=080000004a51c003, pud=080000004a519003, pmd=0000000000000000
Jul 07 02:32:43 nixos kernel: Internal error: Oops: 96000006 [#1] SMP
Jul 07 02:32:43 nixos kernel: Modules linked in: bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev snd_soc_hdmi_codec raspberrypi_cpufreq mc hci_uart snd_bcm2835(C) btqca btsdio btbcm bluetooth brcmfmac brcmutil ip6_tables cfg80211 xt_conntrack nf_conntrack clk_raspberrypi ecdh_generic nf_defrag_ipv6 nf_defrag_ipv4 crct10dif_ce raspberrypi_hwmon broadcom bcm_phy_lib reset_raspberrypi rfkill ecc 8021q genet garp mrp iproc_rng200 i2c_bcm2835 rng_core pwm_bcm2835 xt_tcpudp bcm2711_thermal mdio_bcm_unimac vchiq(C) nvmem_rmem ip6t_rpfilter uio_pdrv_genirq uio ipt_rpfilter xt_pkttype nft_compat nft_counter nf_tables libcrc32c sch_fq_codel nfnetlink zfs(PO) zunicode(PO) zzstd(O) zlua(O) zcommon(PO) znvpair(PO) zavl(PO) icp(PO) spl(O) tap macvlan bridge stp llc fuse ip_tables x_tables xhci_pci xhci_pci_renesas vc4 cec drm_kms_helper drm pcie_brcmstb dm_mod
Jul 07 02:32:43 nixos kernel: CPU: 3 PID: 830 Comm: irq/53-vc4 hdmi Tainted: P         C O      5.15.50 #1-NixOS
Jul 07 02:32:43 nixos kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
Jul 07 02:32:43 nixos kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Jul 07 02:32:43 nixos kernel: pc : vc4_hdmi_enable_scrambling+0x44/0x268 [vc4]
Jul 07 02:32:43 nixos kernel: lr : vc4_hdmi_connector_detect+0x68/0x250 [vc4]
Jul 07 02:32:43 nixos kernel: sp : ffff800008e93980
Jul 07 02:32:43 nixos kernel: x29: ffff800008e93980 x28: 0000000000000010 x27: 0000000000001e00
Jul 07 02:32:43 nixos kernel: x26: 0000000000000002 x25: 0000000000000000 x24: 0000000000000000
Jul 07 02:32:43 nixos kernel: x23: 0000000000000002 x22: ffffd2ae6fe3ac30 x21: 0000000000000001
Jul 07 02:32:43 nixos kernel: x20: ffff322708567080 x19: ffff3227085674f8 x18: 0000000000000000
Jul 07 02:32:43 nixos kernel: x17: 58b0805a70f23000 x16: ffffd2aec012cdd0 x15: 1d0100000183010f
Jul 07 02:32:43 nixos kernel: x14: 06e3ff00e20c0000 x13: 06e3ff00e20c0000 x12: 0fe401c305e30380
Jul 07 02:32:43 nixos kernel: x11: 091300963e10102d x10: 00000000000000b1 x9 : ffffd2ae6fe28b58
Jul 07 02:32:43 nixos kernel: x8 : 000000000000000d x7 : 0000000000000100 x6 : 0000000000000000
Jul 07 02:32:43 nixos kernel: x5 : 0000000000000000 x4 : ffff32273fbdb280 x3 : 00000000000018af
Jul 07 02:32:43 nixos kernel: x2 : 00000000000003e8 x1 : 000000001443fd00 x0 : 0000000000000000
Jul 07 02:32:43 nixos kernel: Call trace:
Jul 07 02:32:43 nixos kernel:  vc4_hdmi_enable_scrambling+0x44/0x268 [vc4]
Jul 07 02:32:43 nixos kernel:  vc4_hdmi_connector_detect+0x68/0x250 [vc4]
Jul 07 02:32:43 nixos kernel:  drm_helper_probe_detect+0xb4/0xe0 [drm_kms_helper]
Jul 07 02:32:43 nixos kernel:  drm_helper_probe_single_connector_modes+0x5f8/0x768 [drm_kms_helper]
Jul 07 02:32:43 nixos kernel:  drm_client_modeset_probe+0x240/0x10c8 [drm]
Jul 07 02:32:43 nixos kernel:  __drm_fb_helper_initial_config_and_unlock+0x50/0x520 [drm_kms_helper]
Jul 07 02:32:43 nixos kernel:  drm_fb_helper_hotplug_event.part.0+0xd8/0xe8 [drm_kms_helper]
Jul 07 02:32:43 nixos kernel:  drm_fbdev_client_hotplug+0x44/0x1c0 [drm_kms_helper]
Jul 07 02:32:43 nixos kernel:  drm_client_dev_hotplug+0x88/0xd8 [drm]
Jul 07 02:32:43 nixos kernel:  drm_kms_helper_hotplug_event+0x3c/0x50 [drm_kms_helper]
Jul 07 02:32:43 nixos kernel:  vc4_hdmi_hpd_irq_thread+0x30/0x40 [vc4]
Jul 07 02:32:43 nixos kernel:  irq_thread_fn+0x34/0xa8
Jul 07 02:32:43 nixos kernel:  irq_thread+0x154/0x2d0
Jul 07 02:32:43 nixos kernel:  kthread+0x128/0x138
Jul 07 02:32:43 nixos kernel:  ret_from_fork+0x10/0x20
Jul 07 02:32:43 nixos kernel: Code: f9402a60 52807d02 529fa001 72a28861 (f941a400) 
Jul 07 02:32:43 nixos kernel: ---[ end trace c7bf4b187d61f972 ]---
Jul 07 02:32:44 nixos kernel: genirq: exiting task "irq/53-vc4 hdmi" (830) is an active IRQ thread (irq 53)
```

Mic92, 2022-07-07:
Here is the function that causes the NULL dereference: https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/vc4/vc4_hdmi.c

Mic92, 2022-07-07:
There have been a few fixes two function. You might want to try the latest linux kernel: https://github.com/torvalds/linux/blame/master/drivers/gpu/drm/vc4/vc4_hdmi.c#L219

nixos-discourse, 2022-07-08:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/nix-raspberry-pi-4-images-dont-work-on-4k-monitor/20140/8