NixOS/nixos-hardware #439: module request: smbios-thermal-ctl

colemickens, 2022-07-30:
see: https://wiki.archlinux.org/title/Dell_XPS_13_(9370)#Thermal_Modes_/_Fan_profiles

It would be nice to have a little module that sets this on boot to a pre-set value. (I use a screen-less XPS 9370 as a battery-backup unifi+home-assistant "server" and would like to always go into performance mode.)

sokai, 2024-01-15:
@colemickens 

FYI:, As long as there isn't such a *module*: I install `libsmbios` and create a systemd service:
```nix
[…]
  environment.systemPackages = with pkgs; [
    libsmbios
  ];

  systemd = {
    services = {
      dell-thermal-mode = {
        after = [ "post-resume.target" ];
        serviceConfig = {
          Type = "oneshot";
          Restart = "no";
          ExecStart = "${pkgs.libsmbios}/bin/smbios-thermal-ctl --set-thermal-mode=quiet";
        };
        wantedBy = [ "multi-user.target" "post-resume.target" ];
      };
    };
  };
[…]
```
All creds are going to: https://github.com/jpas/etc-nixos/blob/88abdbf34bc2dc57c92a8f6fe26501156d9bc04e/profiles/hardware/laptop-dell-xps-13-9300.nix#L83-L98

Lyndeno, 2024-06-28:
FYI, there is a soon to be mainlined module in the kernel for controlling this interface via platform_profiles.

Please see:
https://www.phoronix.com/news/Dell-PC-Driver-For-Linux-6.11

https://git.kernel.org/pub/scm/linux/kernel/git/pdx86/platform-drivers-x86.git/patch/?id=996ad412981024a9bb90991ab195685d37187bbd

Lyndeno, 2024-06-28:
If you use software the changes platform_profile, such as power-profiles-demon, this will affect you.

This patch will also block smbios-thermal-ctl from working, as the kernel is now in control of the fan modes, via platform_profile.

colemickens, 2024-06-28:
Huh, interesting. Better late than never, I suppose.