NixOS/nixos-hardware #443: Lenovo/T14s/AMD: Always sets the kernel to be `pkgs.linuxPackages_latest`

etu, 2022-08-10:
Currently the nixos-hardware for T14s/AMD (https://github.com/NixOS/nixos-hardware/blob/master/lenovo/thinkpad/t14s/amd/gen1/default.nix) it defaults to use of `pkgs.linuxPackages_latest`, this doesn't always work with third party kernel modules such as `zfs` or `VirtualBox`.

So to get my system to build I had to set the kernel myself:
```nix
{
  boot.kernelPackages = pkgs.linuxPackages_5_18;
}
```

However, this didn't work because the nixos-hardware module had already set the kernel to 5.19.

So I had to force it:
```nix
{
  boot.kernelPackages = lib.mkForce pkgs.linuxPackages_5_18;
}
```

Is this really right?

etu, 2022-08-10:
@Mic92 That would be nice, then I could override it without mkForce I guess? :slightly_smiling_face:

Mic92, 2022-09-03:
Yes.