NixOS/nixos-hardware #451: 0BSD license

toastal, 2022-09-01:
https://spdx.org/licenses/0BSD.html

Wouldn't 0BSD license be more appropriate for software than CC0?

Mic92, 2022-09-03:
Maybe but I would say re-licensing sounds too much work given the amount of contributors we would need to task. So I would say it's not worth it.

toastal, 2022-09-04:
I understand. I just recall [reading Fedora](https://www.mail-archive.com/devel@lists.fedoraproject.org/msg183333.html) is now disallowing CC0 software and how that may affect the philosophies of other projects moving forward.

> The reason for the change: Over a long period of time a consensus has been building in FOSS that licenses that preclude any form of patent licensing or patent forbearance cannot be considered FOSS. CC0 has a clause that says: "No trademark or patent rights held by Affirmer are waived, abandoned, surrendered, licensed or otherwise affected by this document." (The trademark side of that clause is nonproblematic from a FOSS licensing norms standpoint.) The regular Creative Commons licenses have similar clauses.

Mic92, 2022-09-15:
I see how this might be a problem in general for software. But specifically for nixos-hardware I think the risk of violating trademarks or patents with hardware quirks and configuration snippets is particular high.

LucasFA, 2024-12-14:
I am unfamiliar with 0BSD, but CC0 is compatible with the MIT license only with certain strings attached, [according to this user](https://forum.openmod.org/t/combined-licensing-with-cc0-1-0-and-mit/4386#:~:text=Note%20carefully%20that%20the%20CC0,not%20additionally%20contain%20patentable%20material).

>"Note carefully that the CC0‑1.0 license is not automatically inbound-compatible with the MIT license. In other words, material under CC0‑1.0 cannot be added to a codebase under MIT unless it can be confirmed that that particular import does not additionally contain patentable material."

This technically causes a bit of a legal issue for nixpkgs with upstreaming  (#992) in the vanishingly unlikely case of a submarine patent. Not a lawyer here.

In any case, continuing to use CC0 opens up this project and Nixpkgs to future contributions not including an assurance of either being patent-free or of providing an implied license to use the patent. It's just the possibility of it that makes this a bit thorny. Bit of a downhill rolling boulder issue, kind of. 

In my limited understanding of the topic, the good news is that dual licensing CC0 OR MIT (or only MIT), wouldn't technically require the permission of contributors, it "only" requires not having any patents. 

The most proper way of making sure is asking every contributor to license to MIT. A more daring way is just saying that there's certainly no patents and change the license, but I get not signing that dotted line.

(PS: anywhere I said MIT could be MIT-0, or possibly 0BSD as long as it also grants patent licenses)