NixOS/nixos-hardware #453: [Request] Add workaround for Skylake and Kabylake R built-in Speaker and Mic Support

LazyGeniusMan, 2022-09-04:
There was a bug on Skylake and Kabylake R laptop where built-in soundcard (ALC256) is not detected[^1][^2][^3][^4]

There is a fix for that bug that require compile something with `alsatplg` and copy the file to `/lib/firmware/` and add a line in `/etc/modprobe.d/*.conf`. Complete steps is here: https://gist.github.com/crojewsk/4e6382bfb0dbfaaf60513174211f29cb#topology

Could someone do this or atleast help me implement this? I read that modprobe can be added in config file but I'm not sure where `/lib/firmware/` is, I can't find the folder in NixOS. When I use other distro I can find the folder and this workaround is working.

[^1]: https://bugzilla.redhat.com/show_bug.cgi?id=1933423
[^2]: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1915117
[^3]: https://bugzilla.kernel.org/show_bug.cgi?id=201251
[^4]: https://patches.linaro.org/project/stable/patch/20210202133001.591879592@linuxfoundation.org/

LazyGeniusMan, 2022-09-12:
I tried with this nix file but it's not working (it was working when I apply it in other distro), anything I missed based on the guide on gist?
```
{ config, pkgs, ... }:

let alsa-topology = pkgs.alsa-topology-conf;
in
{
  # Add necessary packages
  environment.systemPackages = with pkgs; [
    alsa-ucm-conf
    alsa-topology-conf
    alsa-lib
    alsa-utils
  ];

  # Compile ALSA Topology
  hardware.firmware = [
    (pkgs.runCommand "alsa-hda-dsp-topology" { nativeBuildInputs = [ pkgs.alsa-utils ]; } ''
      mkdir -p $out/lib/firmware
      alsatplg -c ${alsa-topology}/share/alsa/topology/hda-dsp/skl_hda_dsp_generic-tplg.conf -o $out/lib/firmware/skl_hda_dsp_generic-tplg.bin
    '')
    pkgs.alsa-firmware
    pkgs.sof-firmware
  ];
  
  # Replace default DSP Firmware
  system.replaceRuntimeDependencies = [
    {
      original = pkgs.firmwareLinuxNonfree;
      replacement = pkgs.firmwareLinuxNonfree.overrideAttrs (super: {
        postInstall = ''
          cd $out/lib/firmware/intel/
          rm dsp_fw_release.bin
          ln -s dsp_fw_kbl_v3420.bin dsp_fw_release.bin
        '';
      });
    }
  ];
  
  # Add modprobe
  boot.extraModprobeConfig = "options snd-intel-dspcfg dsp_driver=1\n";
}
```

Mic92, 2022-09-15:
Have you checked if `/run/current-system/firmware` contains the changes you would expect it to have? An alternative to `replaceRuntimeDependencies` would be to use `nixpkgs.config.packageOverrides` to override `firmwareLinuxNonfree`

closed 2026-10-03 by LazyGeniusMan