NixOS/nixos-hardware #493: Dell XPS 9320 support

notgne2, 2022-10-30:
This laptop is a bit finicky and different to support due to the very weird webcam, although I have hacked something together that makes it possible to use it, even if not properly, and I think getting it cleaned up and turned into a module for this repository would be ideal.

I have uploaded a GitHub Gist of my current hack here https://gist.github.com/notgne2/41ec4ce08982d42d803aac2dc485e27d - and it makes utilizing the webcam through gstreamer possible, also some other hardware related options. There are a few mentions of a broken microphone, though I actually did get it working, I just hadn't updated any of the comments as the NixOS config did not need changing, just running `alsactl init` (I am not sure how many of my microphone-related fixes are responsible for this, and which are redundant).

I think @edolstra has the same laptop too and may be interested in this, if there are any others it might be worth tagging them too. If this issue gets resolved it would make NixOS basically the only GNU/Linux OS to support the XPS 9320 (despite Ubuntu coming pre-installed on some)

notgne2, 2023-01-17:
I've updated the Gist to remove a lot of redundant stuff and also add a helper script for actually running it, which makes it a lot easier to show off/test. I'm not sure if I should hold off on making a PR or not, but it would be nice to get this daemonized somehow and deal with the minor remaining microphone issue (only the left mic works, and you must run `sudo alsactl init` on boot manually).

Daemonization is a bit difficult here because the gstreamer pipeline is not lazy, it will process your video and shine an annoying LED in your face the whole time if you leave it running, I think there is maybe a `v4l2sink` alternative that can turn itself off and on?

Otherwise this should work for anybody stumbling across this I think, just use this module and run `stupid-webcam` when you want to use your webcam.

notgne2, 2023-01-18:
Another update to the Gist, I've added `v4l2-relayd` which now makes the webcam fully automatic (no more stupid script), and updated the pipeline to be compatible with Firefox

Mic92, 2023-02-18:
Just make a PR for it.

closed 2023-02-18 by Mic92

perstarkse, 2023-07-06:
@notgne2 Any update on this? I see there’s passed some months, but a PR would be appreciated if it still works. From what I’ve gathered kernel support is still quite far away for the webcam.

AdrienLemaire, 2025-01-05:
@Mic92 can you re-open this issue ? The related PR was never merged.
As a Dell XPS 9320 user, I still have issues with webcam and mic after installing NixOS.

Mic92, 2025-01-05:
If no one that has the hardware is willing to make the gist to make a pull request than, I don't think there is anything actionable for us as the maintainer of nixos-hardware to re-open this issue. We cannot do this work for you. Sorry.