NixOS/nixos-hardware #503: apply-overlays-dtmerge for Pi4 seems to fail on 22.05

WizardUli, 2022-11-15:
```nix
{
  hardware = {
    raspberry-pi."4".apply-overlays-dtmerge.enable = true;

    deviceTree = {
      overlays = [
        {
          name = "vc4-kms-v3d-pi4";
          dtboFile = "${config.boot.kernelPackages.kernel}/dtbs/overlays/vc4-kms-v3d-pi4.dtbo";
        }
      ];
    };
  };
}
```
where `kernelPackages = pkgs.linuxKernel.packages.linux_rpi4;`
fails with
```
./broadcom -> /nix/store/pc46arc6g6bkx9ny2lsi9hparhizis0q-device-tree-overlays/./broadcom
'./broadcom/bcm2711-rpi-400.dtb' -> '/nix/store/pc46arc6g6bkx9ny2lsi9hparhizis0q-device-tree-overlays/./broadcom/bcm2711-rpi-400.dtb'
'./broadcom/bcm2711-rpi-4-b.dtb' -> '/nix/store/pc46arc6g6bkx9ny2lsi9hparhizis0q-device-tree-overlays/./broadcom/bcm2711-rpi-4-b.dtb'
'./broadcom/bcm2711-rpi-cm4.dtb' -> '/nix/store/pc46arc6g6bkx9ny2lsi9hparhizis0q-device-tree-overlays/./broadcom/bcm2711-rpi-cm4.dtb'
Applying overlay vc4-kms-v3d-pi4 to bcm2711-rpi-400.dtb
cp: cannot create regular file '/nix/store/vrs5pi8qi943svd9l4vmz2bmz984klcj-linux-5.15.32-1.20220331/dtbs/overlays/vc4-kms-v3d-pi4.dtbo.dtbo': Permission denied
```
Is this a regression or am I doing something wrong?
Thanks.

carlossless, 2022-12-17:
Hey @WizardUli, I've reproduced the issue and verified that the fix works in #518, hoping to get it merged soon.

Mic92, 2023-01-24:
close for now, we can re-open if it does not work.

closed 2023-01-24 by Mic92