NixOS/nixos-hardware #504: Fails to build microsoft-surface Kernel

Yeshey, 2022-11-17:
I manage my system with a flake, and included the surface configuration in a similar fashion to the one suggested in the readme ([here is my configuration](https://github.com/Yeshey/nixOS-Config)).

I have a Surface Pro 7, and it would build the kernel fine, but recently, after running `nix flake update --commit-lock-file`, with the new versions of the nixpkgs, it fails to build the surface kernel linux-5.16.11

```
...
  CC [M]  drivers/iio/light/zopt2201.o
  LD [M]  drivers/infiniband/hw/hfi1/hfi1.o
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: warning: arch/x86/power/hibernate_asm_64.o: missing .note.GNU-stack section implies executable stack
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.btf
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: warning: arch/x86/power/hibernate_asm_64.o: missing .note.GNU-stack section implies executable stack
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: warning: .tmp_vmlinux.btf has a LOAD segment with RWX permissions
  BTF     .btf.vmlinux.bin.o
  LD      .tmp_vmlinux.kallsyms1
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: warning: .btf.vmlinux.bin.o: missing .note.GNU-stack section implies executable stack
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: warning: .tmp_vmlinux.kallsyms1 has a LOAD segment with RWX permissions
  KSYMS   .tmp_vmlinux.kallsyms1.S
  AS      .tmp_vmlinux.kallsyms1.S
  LD      .tmp_vmlinux.kallsyms2
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: warning: .btf.vmlinux.bin.o: missing .note.GNU-stack section implies executable stack
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: warning: .tmp_vmlinux.kallsyms2 has a LOAD segment with RWX permissions
  KSYMS   .tmp_vmlinux.kallsyms2.S
  AS      .tmp_vmlinux.kallsyms2.S
  LD      vmlinux
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: warning: .btf.vmlinux.bin.o: missing .note.GNU-stack section implies executable stack
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
/nix/store/039g378vc3pc3dvi9dzdlrd0i4q93qwf-binutils-2.39/bin/ld: warning: vmlinux has a LOAD segment with RWX permissions
  BTFIDS  vmlinux
FAILED: load BTF from vmlinux: Invalid argument
make: *** [../Makefile:1161: vmlinux] Error 255
make: *** Deleting file 'vmlinux'
```

# Compiling computer System Info:

```console
[user@system:~]$ nix-shell -p nix-info --run "nix-info -m"
 - system: `"x86_64-linux"`
 - host os: `Linux 5.15.78, NixOS, 22.11 (Raccoon), 22.11.20221114.013fcdd`
 - multi-user?: `yes`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.11.0`
 - channels(root): `"nixos"`
 - channels(yeshey): `""`
 - nixpkgs: `/nix/var/nix/profiles/per-user/root/channels/nixos`
```

Mic92, 2022-11-17:
cc @kiana-S @mexisme

Yeshey, 2022-11-17:
I'm open if there's anything I can do to help test it, but I'd need rather precise instructions as I'm pretty new to this whole world

mexisme, 2022-11-17:
> I'm open if there's anything I can do to help test it, but I'd need rather precise instructions as I'm pretty new to this whole world

I can take a closer look over the weekend.
I was going to see what the latest kernel supported by upstream was in the next week or so, anyway.

Note that the kernel builds are (were?) substantially the same for all the Surfaces, so will hopefully be a good initial indicator from my machine.

Yeshey, 2022-11-24:
@mexisme any news on this?

mexisme, 2022-11-24:
> @mexisme any news on this?

I'm so sorry, this week was an unexpected mess, so I forgot about it!

I'll try to get to it tomorrow night.

Yeshey, 2022-11-24:
> > @mexisme any news on this?
> 
> I'm so sorry, this week was an unexpected mess, so I forgot about it!
> 
> I'll try to get to it tomorrow night.

Np np :)
Just tag me or the issue when you push so I can test it and hopefully close it

Yeshey, 2022-11-27:
@mexisme I hate to bother you about this again

mexisme, 2022-11-28:
> @mexisme I hate to bother you about this again :sweat_smile:

Don't worry about it.

I can def. replicate it, and my current efforts is to try to upgrade to kernel 5.19.2; some light googling suggests it might be related.
I've targeted 5.19 because I know there have been some reasonably important fixes for some other platforms in 5.19, so was hoping to capture those.
The latest patches from linux-surface seems to be for 5.19.2.

I'm hoping to follow that with a version for the 6.0 series, as experiences on a different Laptop architecture (for AMD) suggests there are some pretty huge improvements there.

mexisme, 2022-11-28:
I may even try to skip straight to the 6.0 series, since that has more fixes.

Yeshey, 2022-11-30:
If there are no major problems, I'd be down for the bleeding edge kernel ¯\_ツ_/¯

But maybe we should even roll back in the meantime as this is kind of a critical issue

christophcharles, 2022-12-04:
I encountered the same issue and bumped the kernel version to 5.19.17. This seems to solve the problem. At least it compiles and the kernel boots fine. See #509

Mic92, 2022-12-04:
I went ahead and merged it until @mexisme has figured out how to upgrade to an even newer kernel.

mexisme, 2022-12-04:
> If there are no major problems, I'd be down for the bleeding edge kernel personally ¯\\\_ツ_/¯
> 
> But maybe we should even roll back in the meantime as this is kind of a critical issue

At some point there was a code-change I hadn't expected, so also did a refactor to accomodate it.

I did a test build of both the linux-surface kernel fork (failer with same error) and the upstream + patches (also failed).

The 6.0 series is not actually bleeding edge any more, but is moving quite fast.

> I encountered the same issue and bumped the kernel version to 5.19.17. This seems to solve the problem. At least it compiles and the kernel boots fine. See #509 

Thanks for that, @christophcharles.

6.0.11 was released yesterday, and patches are already in linux-surface, so I'm going to update and re-test the build.

Yeshey, 2022-12-05:
> I went ahead and merged it until @mexisme has figured out how to upgrade to an even newer kernel.

Can confirm that it now builds fine, will leave closing the issue to you guys :+1:

mexisme, 2022-12-05:
> > I went ahead and merged it until @mexisme has figured out how to upgrade to an even newer kernel.

There are a couple of PRs for that.
Feel free to take a look, if you feel inclined.

> Can confirm that it now builds fine, will leave closing the issue to you guys :+1: 

SGTM.

PR #511 (2022-12-13): Kernel 6.0.11 for Microsoft Surface devices
###### Description of changes

Update to kernel 6.0.11 on Microsoft Surface devices.

Fixes #504.
Depends on #507 (please review and merge that, first).

###### Things done

- [x] Tested the changes in your own NixOS Configuration
- [x] Tested the changes end-to-end by using your fork of `nixos-hardware` and
      importing it via `<nixos-hardware>` or Flake input
closed 2022-12-13 by bors[bot]