NixOS/nixos-hardware #512: acpi_backlight=native introduce issues to my keyboard backlight

baronleonardo, 2022-12-07:
This option

```nix
  boot.kernelParams = [
    # Force use of the thinkpad_acpi driver for backlight control.
    # This allows the backlight save/load systemd service to work.
    "acpi_backlight=native"
  ];
```

from "nixos-hardware/lenovo/thinkpad/e14/default.nix"

currently I have 2 issues related to keyboard backlight
1- the keyboard backlight (on Gnome) doesn't go off when the screensaver start
2- it always enable itself on when I end the screensaver (by moving the cursor)

solution:
if I change to `acpi_backlight=vendor` everything goes well.

Mic92, 2022-12-15:
cc @ebbertd

ebbertd, 2022-12-15:
I no longer have an E14 available to test this.