NixOS/nixos-hardware #521: Fingerprint reader does not work on ThinkPad x270

NicolasHov, 2022-12-20:
Hello everyone,

**TL;DR** The built-in fingerprint reader of the ThinkPad x270 seems to works with the driver [`python-validity`](https://github.com/uunicorn/python-validity) package, but it's not available in `nixpkgs` yet.

I tried to list my devices with:
```
nicolas@nixos ~ % lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 013: ID 138a:0097 Validity Sensors, Inc. 
Bus 001 Device 003: ID 04f2:b5ab Chicony Electronics Co., Ltd Integrated Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I understand that the fingerprint reader is `Bus 001 Device 013: ID 138a:0097 Validity Sensors, Inc. `

And when running `fprint-enroll` it fails with:
```
Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice: No devices available
```

I tried different options but get the same result:

* I read this thread: https://discourse.nixos.org/t/how-to-use-fingerprint-unlocking-how-to-set-up-fprintd-english/21901/2
  I enabled `fprint` in my configuration.nix file and first tried `vfs0090` driver then `goodix` driver but without success in both cases.
  ```
  services.fprintd.enable = true;
  services.fprintd.tod.enable = true;
  services.fprintd.tod.driver = pkgs.libfprint-2-tod1-vfs0090;
  ```

* I read this other thread (for ThinkPad T480): https://discourse.nixos.org/t/thinkpad-t480-fingerprint-reader-support/17448/3 which confirms that there were no `python-validity` package for NixOs.

* I then write a nix script for python-validity to add to my configuration:
  ```nix
  { lib, python3Packages }:
  with python3Packages;
  buildPythonApplication {
    pname = "python-validity";
    version = "0.12";
  
    propagatedBuildInputs = [ cryptography pyusb pyyaml ];
  
    driverPath = /usr/share/python-validity/6_07f_lenovo_mis_qm.xpfwext;
  
    src = ./.;
  }
  ```
but getting the same error as before.

* I also looked on Validity official support, where someone already asked about a Nix driver in 2019 for my device: https://gitter.im/Validity90/Lobby?at=5caf58edf851ee043d9d85d3

Thanks if you have any idea to help me :)

NicolasHov, 2022-12-21:
I also opened an issue in NixOS/nixpkgs/ https://github.com/NixOS/nixpkgs/issues/207116

nixos-discourse, 2022-12-21:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/thinkpad-x270-fingerprint-reader-support/24177/1

ahbnr, 2023-01-01:
I also need python-validity for my T480. I managed to create some nix packages based on the AUR packages and it works for me now.
In total, three packages are needed to get it to work: open-fprintd, fprintd-clients, and python-validity.

However, I am completely new to NixOS and Nix. Hence, I wrote those packages by copying and adapting existing Nix package definitions and applying some guesswork.
Hence, someone more experienced with this than me probably needs to review these packages. Also, I currently load them manually with `callPackage` from local files. I still need to create a nixpkgs fork or a flake git repository to load them properly.

In any case, to get it to work, put the files in the following repository in a folder `packages` relative to your `configuration.nix`: https://github.com/ahbnr/nixos-python-validity

You can then load them in your `configuration.nix` like this:
```nix
...
let
  # load custom packages for driving the fingerprint sensor.
  # This probably conflicts with the default fprintd service, so do not enable services.fprintd
  open-fprintd = (pkgs.callPackage ./packages/open-fprintd/default.nix {});
  fprintd-clients = (pkgs.callPackage ./packages/fprintd-clients/default.nix {});
  python-validity = (pkgs.callPackage ./packages/python-validity/default.nix {});
in
  ...
  environment.systemPackages = with pkgs; [
    ...
    open-fprintd
    fprintd-clients
    python-validity
    ...
  ];
  
  ...
  # Enable services from custom packages
  systemd.packages = [ open-fprintd python-validity ];
  systemd.services.open-fprintd.enable = true;
  systemd.services.python3-validity.enable = true;
  
  # enable fingerprint scanning for sudo
  security.pam.services.sudo.text = ''
    # Account management.
    account required pam_unix.so
    
    # Authentication management.
    auth sufficient pam_unix.so   likeauth try_first_pass nullok
    auth sufficient ${fprintd-clients}/lib/security/pam_fprintd.so
    auth required pam_deny.so
    
    # Password management.
    password sufficient pam_unix.so nullok sha512
    
    # Session management.
    session required pam_env.so conffile=/etc/pam/environment readenv=0
    session required pam_unix.so
  '';
  
  ...
}
```

Then run
```nix
nixos-rebuild test
systemctl start open-fprintd
systemctl start python-validity
```

Now, `fprintd-enroll`, `fprintd-verify` etc. should work. Furthermore, if you do not enter a password in your sudo prompt (just press enter), sudo will authenticate you with your fingerprint instead.

It is a bit worrying, that open-fprintd and fprintd-clients have not received a commit/update in two years. Hence, is there even a chance that they might become official packages?

Mic92, 2023-01-02:
@ahbnr it would be best to just add those python packages to nixpkgs. You already did most of the work already. Than the nixos hardware profile in this repository can be extended as well.

nixos-discourse, 2023-01-18:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/thinkpad-x270-fingerprint-reader-support/24177/3