NixOS/nixos-hardware #522: Surface Laptop 4 AMD: touchpad, keyboard, brightness control not working

tmnvanderberg, 2022-12-21:
Hi  there, and thanks for your work. :)

On updated channel with import `<nixos-hardware/microsoft/surface>`:  keyboard, touchpad and brightness control are not working for me on my new Surface Laptop 4 AMD.

`nixos-rebuild` did cause a kernel compilation, but at this point I'm wondering if it succeeded and if I am on the right kernel, since it's reported as working on the upstream linux-surface repo. Doing another `nixos-rebuild` doesn't change anything.

Best,
Timon

tmnvanderberg, 2022-12-21:
lsmod and dmesg output here: https://gist.github.com/tmnvanderberg/a75b4431472e1348317e9cfb7e71d8e3

Mic92, 2022-12-21:
cc @mexisme

Mic92, 2022-12-21:
There are two kernels two try as well: https://github.com/NixOS/nixos-hardware/pull/511/files#diff-82eaaa50adefad877ba7f54b5a444cf4942804e8d44d474a4fbe3a0cf47c5d5dR12

tmnvanderberg, 2022-12-21:
Sorry, my nix-foo is a little limited and I can't figure out the right 'namespaces' to set `kernelVersion`.
`microsoft-suface.kernelVersion`? `microsoft.surface.kernelVersion`?

mexisme, 2022-12-21:
Hi @tmnvanderberg,
Full disclosure: I'm afraid I don't have access to an AMD Surface, so I may not be able to debug this adequately by myself.
But thanks for dropping `lsmod` and `dmesg`, and I'll take a look at that, now.

In the meantime:

Current `master` of nixos-hardware supports building kernel 6.0.11 and 5.19.17.
The default is currently 6.0.11, mostly because it has quite a few fixes — incl. several for the newer AMD 6000 series CPUs, e.g. internal keyboard  ;-)
You can override the default, and try kernel 5.19.17 by adding `microsoft-surface.kernelVersion = "5.19.17";` to your laptop's `configuration.nix` (or `flake.nix` perhaps).

tmnvanderberg, 2022-12-21:
Thanks both, setting `options.microsoft-surface.kernelVersion = "5.19.17"` atleast fixes the keyboard and trackpad.

mexisme, 2022-12-21:
> Thanks both, setting `microsoft-surface.kernelVersion = "5.19.17"` atleast fixes the keyboard and trackpad.

OK, that sounds like kernel-specific.
Thanks.

mexisme, 2022-12-21:
Have you also tried any of the work-arounds here, @tmnvanderberg ?
- https://github.com/linux-surface/linux-surface/wiki/Surface-Laptop-4#issues-and-workarounds

Especially:
- https://github.com/linux-surface/linux-surface/wiki/Surface-Laptop-4#backlight-control-not-working

You might also like to look at some of the options under https://github.com/NixOS/nixos-hardware/tree/master/common or https://github.com/NixOS/nixos-hardware/tree/master/common/cpu/amd ?

mexisme, 2022-12-21:
I've also realised that the current system has some very Intel-specific settings as that's what was available when I started this, e.g:
```
  boot.extraModprobeConfig = mkDefault ''
    options i915 enable_fbc=1 enable_rc6=1 modeset=1
    options snd_hda_intel power_save=1
    options snd_ac97_codec power_save=1
    options iwlwifi power_save=Y
    options iwldvm force_cam=N
  '';
```
- https://github.com/NixOS/nixos-hardware/blob/master/microsoft/surface/default.nix#L14-L20

I will look into wrapping these into an option, for a start.

But any pointers on a better way to handle these, @Mic92 ?

tmnvanderberg, 2022-12-21:
Indeed, using
```
acpi_backlight=vendor
amd_iommu=off 
iommu=off
```
backlight works.

There are some errors related to IPTSD though:
```
× iptsd.service - IPTSD
     Loaded: loaded (/etc/systemd/system/iptsd.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-12-21 21:01:02 CET; 20ms ago
    Process: 4261 ExecStart=/nix/store/ndz06f142j3nc7avp59g60xmxsrp5jh5-unit-script-iptsd-start/bin/iptsd-start (code=exited, status=254)
   Main PID: 4261 (code=exited, status=254)
         IP: 0B in, 0B out
        CPU: 3ms

dec 21 21:01:02 nixos systemd[1]: Started IPTSD.
dec 21 21:01:02 nixos iptsd-start[4262]: ERROR: ../src/control.c:139: Failed to open /dev/ipts/0: No such file or directory
dec 21 21:01:02 nixos iptsd-start[4262]: ERROR: ../src/main.c:106: Failed to start IPTS: No such file or directory
```

mexisme, 2023-01-09:
I've updated for kernel 6.0.17 in #533 (latest patches from https://github.com/linux-surface/linux-surface) and intend to try a version for 6.1.3.

mexisme, 2023-01-09:
> Indeed, using kernel parameters
> 
> ```
> acpi_backlight=vendor
> amd_iommu=off 
> iommu=off
> ```
> backlight works.

I have a non-MS AMD 6k laptop, and I seem to remember that some kernel versions in the 5.15+ range worked better with `iommu=pt`, however 6.0.* seemed to fix it.
Likewise, there were a large number of fixes for the `amdgpu` module in both the 6.0.* and 6.1.* versions from AMD engineers — which still seem to be coming fast!

 
> There are some errors related to IPTSD though:
>
> ```
> × iptsd.service - IPTSD
>      Loaded: loaded (/etc/systemd/system/iptsd.service; enabled; vendor preset: enabled)
>      Active: failed (Result: [...]

Oh, that looks like the kernel module wasn't loaded?
Can you see the module from `lsmod | grep -i ipts` ?
If not, will it let you load it with `sudo modprobe ipts` ? (I think that's the correct module name, but can't double-check just yet)
Are there any messages in the kernel logs about it?

Also, if you have time, perhaps you could test the updated kernel from #533 ?
I'm in the process of trying it on a Surface Go, but might not get much further until after work.

tmnvanderberg, 2023-01-09:
Regarding the module:
```
(23:12:34) [~/src/]  lsmod | grep -i ipts
(23:12:43) [~/src/  sudo modprobe ipts
[sudo] password for tbe: 
(23:12:54) [~/src/]  lsmod | grep -i ipts
ipts                   16384  0
mei                   155648  1 ipts
```
So I guess it's loading then.

I will try the updated kernel when I have time, but that might not be before the weekend. :)

mexisme, 2023-01-10:
Hi @tmnvanderberg, #537 should be merged shortly, which has a specialisation for Surface Laptop (AMD).
If you could test it and offer feedback on fixing it, then we can add any fixes directly to that, instead of the broader / original `microsoft/surface` profile.

To use it, you should only need import `<nixos-hardware/microsoft/surface/surface-laptop-amd>` (or the Nix Flakes variant) once it's merged.

tmnvanderberg, 2023-01-22:
Hi @mexisme, I gave it a try and looks fine

```
(14:15:18) [~/src/ocaml_exp]  uname -a
Linux nixos 6.0.17 #1-NixOS SMP PREEMPT_DYNAMIC Wed Jan 4 10:26:32 UTC 2023 x86_64 GNU/Linux
```

only thing still not working is touchscreen

mexisme, 2023-01-22:
> Hi @mexisme, I gave it a try and looks fine
> 
> ```
> (14:15:18) [~/src/ocaml_exp]  uname -a
> Linux nixos 6.0.17 #1-NixOS SMP PREEMPT_DYNAMIC Wed Jan 4 10:26:32 UTC 2023 x86_64 GNU/Linux
> ```
> 
> only thing still not working is touchscreen
> 
> Anything specific you'd like to see tested?
> 

Glad to see things are improving!

I'll dig through the linux-surface docs+wiki tonight/tomorrow to see if I can discover what else might be needed.

From memory, the SL3/AMD has a slightly different chipset from SL3/Intel for touchscreen.

mexisme, 2023-01-22:
> Hi @mexisme, I gave it a try and looks fine
> 
> ```
> (14:15:18) [~/src/ocaml_exp]  uname -a
> Linux nixos 6.0.17 #1-NixOS SMP PREEMPT_DYNAMIC Wed Jan 4 10:26:32 UTC 2023 x86_64 GNU/Linux
> ```
> 
> only thing still not working is touchscreen
> 
> Anything specific you'd like to see tested?
> 

FYI:

I have raised a new PR a few days ago for the `6.1.6` kernel+patches — I don't think it's been merged yet, and I expect to replace the `6.0.x` series with this as `6.0.x` has been EOL'd by kernel maintainers.

But, it may be worth trying out of you have time; it may unfortunately take a wee while to rebuild the kernel, of course.

tmnvanderberg, 2023-01-22:
No problem, I'll give it a whirl when it's merged. Thanks for your work!

mexisme, 2023-01-26:
> Do you mean try it now or when merged?

Sorry for the slow reply.

It's merged, and I've tested it successfully on my Surface Go, but the default is still `6.0.17` for the moment:
- https://github.com/NixOS/nixos-hardware/blob/master/microsoft/surface/common/default.nix#L13
```
[...]

    ./surface-control
  ];

  microsoft-surface.kernelVersion = mkDefault "6.0.17";

  # Seems to be required to properly enable S0ix "Modern Standby":
  boot.kernelParams = mkDefault [ "mem_sleep_default=deep" ];

[...]
```

You can change to `6.1.6` by adding the option `microsoft-surface.kernelVersion = "6.1.6";` to your host's config.

tmnvanderberg, 2023-03-15:
Running 6.1.6 now and it looks good. I'll report back if I find anything broken. 

Thanks for your work.

closed 2023-03-15 by Mic92

tmnvanderberg, 2023-03-17:
Hmm, it seems I can't control display brightness on 6.1.6 :(

mexisme, 2023-03-19:
> Hmm, it seems I can't control display brightness on 6.1.6 :(

I need to see what the latest version patched by the Linux-Surface project.
There have been a myriad of fixes in the 6.1.x upstream, and even the early 6.2.x upstream.

viperML, 2023-08-06:
@tmnvanderberg did you get your touchscreen working?

I also have a SL4 (AMD), and last time I tried (many months ago) it wasn't working: https://github.com/linux-surface/linux-surface/issues/902

tmnvanderberg, 2023-08-06:
> @tmnvanderberg did you get your touchscreen working?
> 
> I also have a SL4 (AMD), and last time I tried (many months ago) it wasn't working: [linux-surface/linux-surface#902](https://github.com/linux-surface/linux-surface/issues/902)

Nope.