NixOS/nixos-hardware #541: Support for thinkpad t16 gen1 amd

yashwardhan-gautam, 2023-01-15:
Hello folks, 
Nixos newbie here. I recently got a ThinkPad t16 gen1 AMD, and I've problems recognizing the wifi module with the basic gnome 22.11 image
Processor: AMD Ryzen™ 7 PRO 6850U Processor (2.70 GHz up to 4.70 GHz)
Display: 40.64cms (16) WQXGA (2560 x 1600), IPS, Anti-Glare, Non-Touch, 100% sRGB, 400 Nits
Graphic Card: Integrated Graphics
Wireless : Qualcomm Wi-Fi 6E NFA725A 11AX (2x2) & Bluetooth® 5.1.

Can I use the configuration file for t14gen3 AMD?
The wifi card on the Lenovo site is `NFA725A Wi-Fi 6E 11AX (2x2)`, not sure if the Qualcomm one makes some difference.

Mic92, 2023-01-15:
Potentially you would need the latest kernel:

```
  boot.kernelPackages = pkgs.linuxPackages_latest;
```

We also have nixos installer that use the latest kernel instead of the latest lts.

Mic92, 2023-01-15:
Here an example: https://github.com/NixOS/nixos-hardware/blob/88016c96c3c338aa801695cdd9f186820bcfe4d6/lenovo/thinkpad/p16s/amd/gen1/default.nix#L5