NixOS/nixos-hardware #552: Surface Pro 5 Touch does not work with updated version of IPTS

phpcitizen, 2023-02-09:
Surface pro 5 touch support does not work with the updated version of iptsd 1.0.1

Works with [iptsd 0.5.1](https://github.com/linux-surface/iptsd/releases/tag/v0.5.1)

It updates automatically.

Mic92, 2023-02-18:
cc @mexisme

mexisme, 2023-02-18:
> Surface pro 5 touch support does not work with the updated version of iptsd 1.0.1
> 
> Works with [iptsd 0.5.1](https://github.com/linux-surface/iptsd/releases/tag/v0.5.1)
> 
> It updates automatically.
> 
> 
> **Edit:** 
> The error is DEVICE not found in the service.
> 
> I created another service:
> ```
> let
>   # For iptsd 0.5.1
>   pkg-0_5_1 = import
>     (builtins.fetchTarball {
>       url = "https://github.com/NixOS/nixpkgs/archive/ee01de29d2f58d56b1be4ae24c24bd91c5380cea.tar.gz";
>       # Hash obtained using `nix-prefetch-url --unpack <url>`
> 
>       sha256 = "0829fqp43cp2ck56jympn5kk8ssjsyy993nsp0fjrnhi265hqps7";
>     })
>     {
>       system = "x86_64-linux";
>     };
> in
> {
>   systemd.services.iptsd2 = {
>     description = "IPTSD";
>     path = [
>       pkg-0_5_1.iptsd
>     ];
>     script = "iptsd";
>     wantedBy = [ "multi-user.target" ];
>   };
> ```
> and it works.
> 
>  I also realized the microsoft-surface.ipts.enable = false; option does not work as it seems to be config is already hard-coded to true. I don't know the case for microsoft-surface.surface-control.enable.
> 

Hardcoded is def. unhelpful.
I'll try to take a look early this coming week (juat flying, atm...)/

phpcitizen, 2023-02-18:
Forgot to mention that the ipts version issue happens in unstable channel.

dotlambda, 2023-02-22:
It's sufficient to add `iptsd` to `services.udev.packages` and `systemd.packages`.

phpcitizen, 2023-02-22:
Will not unstable branch still will fetch the version of iptsd 1.0.1?

dotlambda, 2023-02-22:
> Will not unstable branch still will fetch the version of iptsd 1.0.1?

Yes, and that version (and all later ones) ship with the appropriate udev rule and systemd version.
You won't need `systemd,l.services.iptsd` anymore.

mexisme, 2023-03-02:
So, finally finding time to follow this up.

I'm not familiar with the Meson builder, @dotlambda.

But from https://github.com/linux-surface/iptsd, it looks like it's installing a systemd config, and the [Nixpkgs file](https://github.com/NixOS/nixpkgs/blob/nixpkgs-unstable/pkgs/applications/misc/iptsd/default.nix#L46-L51) fixes the install paths?

Therefore it probably makes sense to add a check into (nixos-hardware/.../ipts)[https://github.com/NixOS/nixos-hardware/blob/77de4cd09db4dbee9551ed2853cfcf113d7dc5ce/microsoft/surface/common/ipts/default.nix] to only be available for v0.5.1 and earlier?

mexisme, 2023-03-02:
> Will not unstable branch still will fetch the version of iptsd 1.0.1?

Looks like it's already [updated to 1.1.1](https://github.com/NixOS/nixpkgs/blob/nixpkgs-unstable/pkgs/applications/misc/iptsd/default.nix#L18) @phpcitizen   ;-)

dotlambda, 2023-03-02:
> It's sufficient to add `pkgs.iptsd` to `services.udev.packages` and `systemd.packages`.

Just make `microsoft-surface.ipts` do this if the iptsd version is at least `1.0`.

mexisme, 2023-03-02:
> > It's sufficient to add `pkgs.iptsd` to `services.udev.packages` and `systemd.packages`.
> 
> Just make `microsoft-surface.ipts.enable = true` do this if the iptsd version is at least `1.0`.

I think you meant that to disable the nixos-hardware systemd config for iptsd if it's at-least `1.0`?
As the nixpkgs version provides `systemd` ?

I quickly knocked together a commit that has an assertion checking older than `1.0.1` as that's what was mentioned above.

However, in retrospect, what we want is to add the Nixpkgs `udev` and `systemd` to `services.udev.packages` and `systemd.packages` for 1.0+, and the local version for everything earlier, yes?

dotlambda, 2023-03-02:
> However, in retrospect, what we want is to add the Nixpkgs `udev` and `systemd` to `services.udev.packages` and `systemd.packages` for 1.0+, and the local version for everything earlier, yes?

Isn't that exactly what I said?

mexisme, 2023-03-02:
> > However, in retrospect, what we want is to add the Nixpkgs `udev` and `systemd` to `services.udev.packages` and `systemd.packages` for 1.0+, and the local version for everything earlier, yes?
> 
> Isn't that exactly what I said?

Yes you did, but the higher-level comment didn't show when I read it via the mobile app.
I only saw it when I reopened in my laptop.

mexisme, 2023-03-19:
Hi all.
Real Life™ delayed me, but I have a draft PR for this:
- https://github.com/NixOS/nixos-hardware/pull/584#issuecomment-1475390206

mexisme, 2023-05-14:
> Hi, thanks for working on this.
> 
> I had the same systemd issue on my Intel Surface Pro 4. I tested the draft pull request and that at least lets me run `nixos-rebuild switch` successfully without activation errors. `microsoft-surface.ipts.enable = true;` doesn't seem to have any effect but that may be by design? In any case, It's easy enough to add the systemd service suggested in the original post and everything works after that.

OK, good to know;  I must've made a mistake in my code.
However, perhaps this patch can be deprecated.
It would be much better to use the upstream version, of course.

fdgefgd, 2023-06-30:
in the end, what was the solution for this issue?

dotlambda, 2023-09-19:
> in the end, what was the solution for this issue?

This might help: https://github.com/NixOS/nixpkgs/issues/240612#issuecomment-1635859341