NixOS/nixos-hardware #607: nvidia/disable.nix isn't working as expected

katexochen, 2023-04-16:
I wanted to disable my Nvidia GPU as chromium wasn't happy with it on wayland. I've added the
[common/gpu/nvidia/disable.nix](https://github.com/NixOS/nixos-hardware/blob/3006d2860a6ed5e01b0c3e7ffb730e9b293116e2/common/gpu/nvidia/disable.nix) to my config, but it didn't disable the device. The udev rules weren't loaded and nouveau was still running. I had to remove all `lib.mkDefault` from the file to actually get the GPU disabled. I don't think that's the expected behavior, is it?

You can find my full issue with chromium, debugging output and solution in [this discourse thread](https://discourse.nixos.org/t/chromium-unstartable-hangs-on-sway-with-nvidia/27294).

Lyndeno, 2023-08-08:
I am also having this issue, on my XPS 9560. module settings and udev rules do not seem to take effect.

What is strange is that this used to work.

Do we want to continue using `lib.mkDefault`? If we use this my understanding is if we wanted to say, blacklist an additional module in our system config, the list here would be completely overwritten.

Lyndeno, 2023-08-08:
Reviewing [CONTRIBUTING.md](https://github.com/NixOS/nixos-hardware/blob/a2dab34/CONTRIBUTING.md), it seems that some of the options should be adjusted to not use `lib.mkDefault` e.g. options that should be merged with user config like lists, sets, and multiline strings. I will create PR(s) to make the necessary changes.

PR #692 (2023-08-09): common/nvidia disable: Remove `lib.mkDefault`
###### Description of changes

Adjust Nvidia disable module to match with CONTRIBUTING.md guidelines. Specifically to remove lib.mkDefault on options we would want to allow merging user config with.

Another PR performing this change treewide may be beneficial.

Related to #691 

Fixes #607 

###### Things done

- [x] Tested the changes in your own NixOS Configuration
- [x] Tested the changes end-to-end by using your fork of `nixos-hardware` and
      importing it via `<nixos-hardware>` or Flake input
closed 2023-08-09 by Mic92