NixOS/nixos-hardware #611: error while building on surface pro 6

aammfe, 2023-04-27:
I'm facing the following error while building on surface pro 6

```
imports =
    [ # Include the results of the hardware scan.
      ./hardware-configuration.nix
      <nixos-hardware/microsoft/surface/surface-pro-intel>
    ];
system.stateVersion = "22.11";
```

```
source root is linux-6.0.17
source root is linux-6.0.17
setting SOURCE_DATE_EPOCH to timestamp 1672827992 of file linux-6.0.17/virt/lib/irqbypass.c
patching sources
configuring
no configure script, doing nothing
building
build flags: -j8 SHELL=/nix/store/kga2r02rmyxl14sg96nxbdhifq3rb8lc-bash-5.1-p16/bin/bash CROSS= CC=cc LD=cc
  CC       lib/cpufreq.o
  CC       lib/cpupower.o
  CC       lib/cpuidle.o
  CC       lib/acpi_cppc.o
  CC       utils/helpers/amd.o
  CC       utils/helpers/msr.o
  CC       utils/helpers/sysfs.o
  CC       utils/helpers/misc.o
  CC       utils/helpers/cpuid.o
  CC       utils/helpers/pci.o
  CC       utils/helpers/bitmask.o
  CC       utils/idle_monitor/nhm_idle.o
  CC       utils/idle_monitor/snb_idle.o
  CC       utils/idle_monitor/hsw_ext_idle.o
  CC       utils/idle_monitor/amd_fam14h_idle.o
  CC       utils/idle_monitor/cpuidle_sysfs.o
  CC       utils/idle_monitor/mperf_monitor.o
  CC       utils/idle_monitor/cpupower-monitor.o
  CC       utils/cpupower.o
  CC       utils/cpufreq-info.o
  CC       utils/cpufreq-set.o
  CC       utils/cpupower-set.o
  CC       utils/cpupower-info.o
  CC       utils/cpuidle-info.o
  CC       utils/cpuidle-set.o
  MSGFMT   po/de.gmo
  MSGFMT   po/fr.gmo
  MSGFMT   po/it.gmo
  MSGFMT   po/cs.gmo
  MSGFMT   po/pt.gmo
  LD       libcpupower.so.0.0.1
make[1]: Entering directory '/build/linux-6.0.17/tools/power/cpupower/bench'
  CC       main.o
  CC       parse.o
  CC       system.o
  CC       benchmark.o
  CC       cpupower
setting SOURCE_DATE_EPOCH to timestamp 1672827992 of file linux-6.0.17/virt/lib/irqbypass.c
patching sources
applying patch /nix/store/mibwk9p63p9pv0ps54s93ycyav20g1k2-source/patches/6.0/0001-surface3-oemb.patch
/nix/store/0jmdsgfnd6aakxdr0sl5l7zzfs59hdrw-stdenv-linux/setup: line 1002: /nix/store/mibwk9p63p9pv0ps54s93ycyav20g1k2-source/patches/6.0/0001-surface3-oemb.patch: No such file or directory
error: builder for '/nix/store/m9viqf7ldlvya51z9dy0pzfrcffm0lii-linux-config-6.0.17.drv' failed with exit code 1
error: 1 dependencies of derivation '/nix/store/fxrk2w4z93xbibryvj3m83fl66a2595k-linux-6.0.17.drv' failed to build
error: 1 dependencies of derivation '/nix/store/0qlcp0n89mzmy1fpn0nr7ky1wvd14ipj-nixos-system-nixos-22.11.3935.60c0f762658.drv' failed to build
```

aammfe, 2023-05-03:
I have also tried to use the fixed version of nixos-hardware but getting same error

```
imports =
    [ # Include the results of the hardware scan.
      ./hardware-configuration.nix
      "${builtins.fetchGit { url = "https://github.com/NixOS/nixos-hardware?rev=62e8ddb93f08487f63dbb339d1e572b214f81d9"; }}/microsoft/surface/surface-pro-intel"
    ];
system.stateVersion = "22.11";
```

```
patching sources
applying patch /nix/store/mibwk9p63p9pv0ps54s93ycyav20g1k2-source/patches/6.0/0001-surface3-oemb.patch
/nix/store/0jmdsgfnd6aakxdr0sl5l7zzfs59hdrw-stdenv-linux/setup: line 1002: /nix/store/mibwk9p63p9pv0ps54s93ycyav20g1k2-source/patches/6.0/0001-surface3-oemb.patch: No such file or directory
error: builder for '/nix/store/m9viqf7ldlvya51z9dy0pzfrcffm0lii-linux-config-6.0.17.drv' failed with exit code 1
error: 1 dependencies of derivation '/nix/store/fxrk2w4z93xbibryvj3m83fl66a2595k-linux-6.0.17.drv' failed to build
error: 1 dependencies of derivation '/nix/store/vjgmbwjnx9m0ya8jhzjlmyvx69nl43b9-nixos-system-nixos-22.11.3935.60c0f762658.drv' failed to build
```

aammfe, 2023-05-03:
@mexisme

NobbZ, 2023-05-06:
I looked into upstream repo, the patches seem to be only provided for 6.1 and 6.2 kernels there.

https://github.com/linux-surface/linux-surface/tree/efe7ddadd4aad3d1a6531334fe086931ec8640a1/patches

digitizer0, 2023-05-09:
Had the same issue on my Surface 4 Pro.

I added the following row to the configuration.nix
`microsoft-surface.kernelVersion = "6.1.18";`

There are 3 possible kernels to use.

https://github.com/NixOS/nixos-hardware/tree/master/microsoft/surface/common/kernel

aammfe, 2023-05-09:
thanks for pointing out

fdgefgd, 2023-06-29:
I have the same problem. How can I fix?

ztr-pen, 2023-07-06:
> I have the same problem. How can I fix?

[fdgefgd](https://github.com/fdgefgd) see reply from [digitizer0](https://github.com/digitizer0) on May 9th above.

fdgefgd, 2023-07-07:
> > I have the same problem. How can I fix?
> 
> [fdgefgd](https://github.com/fdgefgd) see reply from [digitizer0](https://github.com/digitizer0) on May 9th above.

I have done that, and the kernel works, however I keep getting the error that iptsd could not find service file now