NixOS/nixos-hardware #616: Odroid HC4 fancontrol fails on 5.15 kernel

metiulekm, 2023-05-01:
When using the HC4 module together with the 5.15 kernel, fancontrol returns the following error:

    Loading configuration from /nix/store/0bd4nbvdvw2nlnhw1p3lz0nn0mj1za7q-fancontrol.conf ...
    Common settings:
      INTERVAL=10
    Settings for hwmon2/pwm1:
      Depends on hwmon0/temp1_input
      Controls hwmon2/fan1_input
      MINTEMP=50
      MAXTEMP=60
      MINSTART=20
      MINSTOP=28
      MINPWM=0
      MAXPWM=255
      AVERAGE=1
    Error: file hwmon2/fan1_input doesn't exist
    At least one referenced file is missing. Either some required kernel
    modules haven't been loaded, or your configuration file is outdated.
    In the latter case, you should run pwmconfig again.

<https://www.armbian.com/odroid-hc4/>, from where the fancontrol configuration comes, has the following note:

> With kernel 5.15.y -> you need to remove FCFANS=hwmon2/pwm1=hwmon2/fan1_input in /etc/fancontrol

This change, applied to the fancontrol Nix config, indeed makes the fancontrol run on my config. It is unclear on which kernel versions this is needed, though it's probably not only 5.15, but maybe 5.15 or newer (or maybe 5.15 or older?); I did not have time to check this yet in any way.

metiulekm, 2023-05-01:
Here's a commit which removes the line on 5.15 or newer: https://github.com/metiulekm/nixos-hardware/tree/hc4-fancontrol-fix

But again I don't know if this should be 5.15 or newer, or 5.15 or older, or maybe exactly 5.15. Hence no PR yet. If you prefer, I can submit a PR which only removes the line on exactly 5.15 (just like the source says).

Mic92, 2023-05-01:
looks good, just open a PR and use `lib.optionalString` instead of the if clause.

metiulekm, 2023-05-02:
I wanted to check on 6.2, but then my HC4 fell into restart loop :/ I don't feel like debuging this now.

I guess I will push a version tomorrow that only removes the line on 5.15, and also do what you @Mic92 said.

commit adab6fd8 (2023-05-05): hardkernel/odroid-hc4: fix fancontrol on 5.15 kernel

Source for the fancontrol config mentions that one of the lines should
be deleted for the 5.15 kernel.

Fixes #616.
PR #618 (2023-05-05): hardkernel/odroid-hc4: fix fancontrol on 5.15 kernel
###### Description of changes

Source for the fancontrol config mentions that one of the lines should be deleted for the 5.15 kernel.

Fixes #616.

###### Things done

- [x] Tested the changes in your own NixOS Configuration
- [x] Tested the changes end-to-end by using your fork of `nixos-hardware` and
      importing it via `<nixos-hardware>` or Flake input

I tested on 5.15 only, sadly couldn't test other versions. I verified that `fancontrol` runs successfully and also that its config file is as expected. Also ran `./tests/run.py '<nixos- [... cut]
closed 2023-05-05 by bors[bot]