NixOS/nixos-hardware #635: Pinebook Pro with LUKS hanging on boot

WesleyAC, 2023-05-25:
I'm having the problem described in https://github.com/samueldr/wip-pinebook-pro/issues/37:

I have a Pinebook Pro. I installed NixOS on it and am using nixos-hardware. I have it partitioned with LUKS, approximately as described [here](https://gist.github.com/martijnvermaat/76f2e24d0239470dd71050358b4d5134). I'm booting using Tow-Boot flashed to the SPI chip.

When I boot, I get two lines of output (as described in the linked issue), the second being:

```
OF: graph: no port node found in /i2c@ff3d0000/fusb30x@22
```

I've tried this both with the default kernel and with the patched 5.16 kernel, both fail. I'm compiling the 6.3 kernel now with the current Manjaro patches to see if that works.

Notably, the error does not seem to be the root cause of the hang, because I see the error if I look in dmesg during the booting of the standard USB install media. I think it's just the last thing that gets printed to the screen before it happens to hang for a unrelated reason. I'm also enabling `boot.crashDump` during this compilation of the kernel to see if that catches anything.

samueldr, 2023-05-25:
Indeed this is not why it is "hanging". That error/warning(?) is related to the power management IC, `fusb30x` is the family of the power IC used here.

Can you try appending `console=tty0` as the last command line argument? You can do this imperatively once at the bootloader if you're using e.g. GRUB (systemd-boot too I think?). I think what is happening here is that the "stdout" of `init` and child processes is sent to another console, which would be `ttyS2` by default on the platform. There is no console multiplexing in the shell-script-based NixOS stage-1, so it ends-up being at the stdout for `init`.

The stdout for `init` is defined by the "last valid console parameter", and when there is none added, it ends-up being "guessed by the hardware configuration", and on ARM+DTB many times it ends-up being a serial console.

So I suspect heavily that everything is working, but working on the serial console instead of fbcon.

WesleyAC, 2023-05-25:
@samueldr indeed, you're spot on. Adding that parameter got the output going the right place, and lets me type in my password to get to stage 2.

Is there a good way to set this parameter to be the default in nixos-hardware?

samueldr, 2023-05-26:
Assuming that previously there were none set, using *`mkBefore`* to add `console=tty0` would be appropriate enough I think. This allows other users to override the console by default without forcing them to `mkAfter`.

Do add a comment explaining why it is necessary, if you do!