NixOS/nixos-hardware #637: Nix build fails for StarFive Visionfive v2 (v1.3b 8GB)

skyrgobbler, 2023-05-27:
Building the example flake as in the example (configured with the v1.3b 8GB patch) fails to build with:

```
error: builder for '/nix/store/75hlaz682g87cybpvaz0qp07rwvprap6-xfsprogs-riscv64-unknown-linux-gnu-6.3.0.drv' failed with exit code 2;
       last 10 log lines:
       >     [CC]     workqueue.lo
       > In file included from randbytes.c:9:
       > ../include/platform_defs.h:27:10: fatal error: urcu.h: No such file or directory
       >    27 | #include <urcu.h>
       >       |          ^~~~~~~~
       > compilation terminated.
       > make[2]: *** [Makefile:77: crc32selftest] Error 1
       > make[2]: *** Waiting for unfinished jobs....
       > make[1]: *** [include/buildrules:36: libfrog] Error 2
       > make: *** [Makefile:84: default] Error 2
       For full logs, run 'nix log /nix/store/75hlaz682g87cybpvaz0qp07rwvprap6-xfsprogs-riscv64-unknown-linux-gnu-6.3.0.drv'.
error (ignored): error: cannot unlink '/tmp/nix-build-texinfo-interactive-riscv64-unknown-linux-gnu-7.0.3.drv-1/texinfo-7.0.3': Directory not empty
error (ignored): error: cannot unlink '/tmp/nix-build-groff-riscv64-unknown-linux-gnu-1.22.4.drv-1/groff-1.22.4': Directory not empty
error: 1 dependencies of derivation '/nix/store/y0kgks00xq0wqkqqxxjwcp6nd52kn58q-etc-system.conf.drv' failed to build
error: 1 dependencies of derivation '/nix/store/y8sfxyqg4mz0f01vi4q3845h1vqjf8sq-system-path.drv' failed to build
error: 1 dependencies of derivation '/nix/store/fz3z0bvyj5icsqrcr30rvi388yf8i3zl-nixos-system-nixos-23.11pre-git.drv' failed to build
error: 1 dependencies of derivation '/nix/store/8mlawsghyaa7ic50rxyvb30paf7fmjgv-ext4-fs.img-riscv64-unknown-linux-gnu.drv' failed to build
error: 1 dependencies of derivation '/nix/store/g25032k5pxrqkmwfvd8da54i3rsqbnwr-nixos-sd-image-23.11pre-git-riscv64-linux-starfive-visionfive2.img-riscv64-unknown-linux-gnu.drv' failed to build
```

leifhelm, 2023-05-30:
I was able to build it with nixpkgs 963006aab35e3e8ebbf6052b6bf4ea712fdd3c28.

tomcharnock, 2023-05-30:
Yes, the device tree overlay (called 8GB patch in the example :)). 

I rebuilt with the revision that you mentioned and the build went smoothly, but after writing the image to an SD card there is no boot on the SD setting. Does the image include the necessary u-boot partition?

leifhelm, 2023-05-31:
The image has the spl and u-boot partitions. It should also boot from emmc (You might have to update the bootloader though). Have you tried all switch configurations? Does the official image boot from sd?

skyrgobbler, 2023-05-31:
Great, that revision works. My VisionFive 2 is now nice and hot and busy compiling everything under the sun. I wonder (but not strongly enough to go and look) what the changes have been to get it to build.

Also, just to note in this thread. When rebooting I sometimes still get the `BOOT fail,Error is 0xffffffff` error a few times in a row and then it just boots up fine.

Interestingly, I don't have any HDMI output only serial, even though it works fine on the Debian image... However, I think we can probably close this issue as that is a different problem?

leifhelm, 2023-06-03:
It is expected that HDMI is not supported in the current kernel version

skyrgobbler, 2023-08-02:
VisionFive 2 now building and working.

closed 2023-08-02 by skyrgobbler