NixOS/nixos-hardware #640: rpi4 on compressed BTRFS root won't boot

n8henrie, 2023-05-31:
Hello,

I have a little project at https://github.com/n8henrie/nixos-btrfs-pi that builds an sdImage for a Pi on a compressed BTRFS rootfs. It's been working great on a Pi3 for a while and I recently decided to try to get it working for a Pi4.

It seems like this repo is the recommended way to get much of the hardware support working, but unfortunately when I enable its module I get a non-booting system (it gets stuck at `waiting for device /dev/disk/by-label/NIXOS_SD ........`).

I'm booting with a custom uboot that I've created via an overlay that enables BTRFS and ZSTD support, I don't know if that is causing the issue?

Perhaps by coincidence, I get the same boot failure (without this module) unless I manually include:

```
boot.initrd.availableKernelModules = [
        "uas"
        "pcie-brcmstb"
        "reset-raspberrypi"
      ];
```

but it looks like these are also included in the nixos-hardware module, so that may not be relevant.

Perhaps it's due to setting `kernelPackages` to `pkgs.linuxKernel.packages.linux_rpi4`? Manually setting just that option seems to break my config as well.

n8henrie, 2023-08-05:
It's not clear what the issue was. A fresh installation with the same boot setup worked without issue; `btrfs check` revealed warnings: about being "not nodesize aligned" which "may cause problem for 64K page system". I suspect this was from nixos-lustrating from an older Raspbian (that had done the initial `mkfs.btrfs` steps with an older kernel).

Regardless, doesn't seem to have been an issue with this repo.

closed 2023-08-05 by n8henrie

nixos-discourse, 2023-08-05:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/rpi4-cant-usb-boot-from-btrfs-with-rpi-kernel/28668/2