NixOS/nixos-hardware #641: Lichee Pi4a support

colemickens, 2023-06-01:
Hello fellow riscv64-linux'ers:

We have visionfive2 support in this repo already. I might have purchased a Lichee Pi4a upon finding out it's faster than the VisionFive2. Of course, shipping will take time but I'm starting to think about nixos-hardware support for it.

Is anyone interested? Maybe has a head-start?

colemickens, 2023-06-01:
Found this on reddit: https://gitee.com/thead-yocto/xuantie-yocto

>It’s a 5.10 kernel with many closed source drivers/blobs for GPU/AI/DSP and using T-Head’s 10.2 GCC.

src: https://www.reddit.com/r/RISCV/comments/12murs3/comment/jgdtz2e/?utm_source=reddit&utm_medium=web2x&context=3

colemickens, 2023-06-19:
This looks active anyway:
* kernel: https://github.com/revyos/thead-kernel/tree/lpi4a
* u-boot: https://github.com/revyos/thead-u-boot/tree/lpi4a
* opensbi: https://github.com/revyos/thead-opensbi/tree/lpi4a

ryan4yin, 2023-08-06:
@colemickens I just made the first NixOS image that can run on Lichee Pi 4A:

https://github.com/ryan4yin/nixos-licheepi4a

There are still a lot of works to do though(: