NixOS/nixos-hardware #651: Pi 4: nixos-install fails with "The option `hardware.i2c' does not exist."

mtlynch, 2023-06-18:
Newcomer to Nix here. I'm trying to install on a Raspberry Pi 4 based on the [nix.dev tutorial](https://nix.dev/tutorials/nixos/build-and-deploy/installing-nixos-on-a-raspberry-pi). The tutorial refers to commit 936e4649098d6a5e0762058cb7687be1b2d90550 of this repo, but I wanted to try with the latest and greatest.

When I try, basing it on `master`, I get this error:

```bash
$ sudo nixos-install --root /
error: The option `hardware.i2c' does not exist. Definition values:
- In `/microsd/nix/store/rfscnzf5wzznazg05n3xvz7rgyln5w7k-source/raspberry-pi/4/i2c1.nix':
    {
      _type = "if";
      condition = true;
      content = {
        enable = true;
    ...
(use '--show-trace' to show detailed location information)
```

* `configuration.nix`: https://gist.github.com/mtlynch/51bc288b8f91ec601dd9163aedda5a05

I bisected the commits and found the issue was introduced in 1bc731fde4af337134331572889de8dd8dbff897. If I try the same `configuration.nix` specifying the preceding commit (ad1114ee372a52aa0b4934f72835bd14a212a642), everything works.

Mic92, 2023-07-10:
cc @domenkozar

domenkozar, 2023-07-10:
cc @NixOS/documentation-team

proofconstruction, 2023-07-11:
Hey @mtlynch, thanks for bringing this to our attention. The `nixos-hardware` module actually isn't necessary anymore unless you need support for specific hardware like the PoE hat; documentation could definitely be better here. I've updated the nix.dev documentation accordingly.

mtlynch, 2023-08-18:
This is now working as a consequence of https://github.com/nixos/nixos-hardware/issues/631

I've updated my tutorial to use the latest version of NixOS (23.11pre) with a working configuration for NixOS on Pi 4, including graphics acceleration:

https://mtlynch.io/nixos-pi4/

closed 2023-08-18 by mtlynch