NixOS/nixos-hardware #655: rpi4 nixos-install error with device tree overlay

EdibleShelfXII, 2023-06-21:
Attempting to build nixos on raspberry pi 4B yields this error:
```
Applying overlay rpi4-vc4-fkms-v3d-overlay to bcm2711-rpi-cm4-io.dtb...
Failed to apply '/nix/store/z1a6cb0ny1zipvn3156q0w1spc8nsi34-rpi4-vc4-fkms-v3d-overlay-dtbo': FDT_ERR_NOTFOUND
error: builder for '/nix/store/2cbab0g3czcg370dc456zklc04h6bmap-device-tree-overlays.drv' failed with exit code 1
error: 1 dependencies of derivation '/nix/store/nxs1rq9c2m5qg3b7rskdqrkij7516wqr-nixos-system-lrivera-nixos-23.11pre495256.04af42f3b31.drv' failed to build
```
Using this configuration:
```
  { config, pkgs, lib, ... }:

  let
    user = "lrivera";
    password = "notmyrealpassword";
    SSID = "mywifi";
    SSIDpassword = "mypassword";
    interface = "wlan0";
    hostname = "lrivera-nixos";
  in {
    imports = ["${fetchTarball "https://github.com/NixOS/nixos-hardware/archive/936e4649098d6a5e0762058cb7687be1b2d90550.tar.gz" }/raspberry-pi/4"];

    fileSystems = {
      "/" = {
        device = "/dev/disk/by-label/NIXOS_SD";
        fsType = "ext4";
        options = [ "noatime" ];
      };
    };

    networking = {
      hostName = hostname;
      networkmanager = {
        enable = true;
      };
    };

    environment.systemPackages = with pkgs; [ vim ];

    services.openssh.enable = true;

    users = {
      mutableUsers = false;
      users."${user}" = {
        isNormalUser = true;
        password = password;
        extraGroups = [ "wheel" ];
      };
    };

    # Enable GPU acceleration
    hardware.raspberry-pi."4".fkms-3d.enable = true;

    services.xserver = {
      enable = true;
      displayManager.lightdm.enable = true;
      desktopManager.xfce.enable = true;
    };

    hardware.pulseaudio.enable = true;
  }
```
modified from the configuration on this page:
https://nix.dev/tutorials/nixos/build-and-deploy/installing-nixos-on-a-raspberry-pi

GiulioCocconi, 2023-06-25:
Related to #631

purcell, 2023-08-09:
Duplicate of #631, indeed, can be closed by the submitter or an admin in favour of the earlier ticket. (Also, note there that we are approaching a resolution.)

closed 2023-12-13 by Mic92