NixOS/nixos-hardware #658: NixOS `<nixos-hardware/microsoft/surface/surface-pro-intel>` does not function on surface pro 7

fdgefgd, 2023-06-29:
I install and then boot up a new instance of nixOS, and wish to add the proper firmware for my hardware (surface pro 7, 8GB RAM, 256GB Storage). I did the following steps:

1. Use and update the nixOS-hardware channel
2. Paste `<nixos-hardware/microsoft/surface/surface-pro-intel>` into my `imports` section
3. Enter the command `sudo nixos-rebuild switch`

Nothing else has been done to the install, I have just installed the system 30 minutes ago. 

It fails to build, and returns the error:
```
error: builder for '/nix/store/0r52xqjn7qv8qc7xx3004qrp7bz8d17c-linux-config-6.0.17.drv' failed with exit code 1
error: 1 dependencies of derivation '/nix/store/hbs2d7kq010n0j00jrijxwiirh936z85-linux-6.0.17.drv' failed to build
error: 1 dependencies of derivation '/nix/store/gfanvy0kp16mqicmgxrsycmnn7mqkbkm-nixos-system-nixos-22.11.4726.6610eb320ef.drv' failed to build
```

I can provide additional information if needed. How can I fix this issue and get the correct surface pro firmware and kernel to function on my nixos installation?

fdgefgd, 2023-06-29:
I have fixed this specific error by putting `microsoft-surface.kernelVersion = "6.1.18";` into my config file. Can we please make this default to avoid the same error in the future, and also add the newest kernel version as well.

Now I am getting a different error:
```
setting up tmpfiles
starting the following units: systemd-udevd-control.socket, systemd-udevd-kernel.socket
warning: the following units failed: iptsd.service

× iptsd.service - IPTSD
     Loaded: loaded (/etc/systemd/system/iptsd.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2023-06-29 22:41:08 BST; 84ms ago
   Duration: 4ms
    Process: 5907 ExecStart=/nix/store/j698k9nl7idlx4rxpm1wa755fqabb436-unit-script-iptsd-start/bin/iptsd-start (code=exited, status=254)
   Main PID: 5907 (code=exited, status=254)
         IP: 0B in, 0B out
        CPU: 3ms

Jun 29 22:41:08 nixos systemd[1]: Started IPTSD.
Jun 29 22:41:08 nixos iptsd-start[5908]: ERROR: ../src/control.c:139: Failed to open /dev/ipts/0: No such file or directory
Jun 29 22:41:08 nixos iptsd-start[5908]: ERROR: ../src/main.c:106: Failed to start IPTS: No such file or directory
Jun 29 22:41:08 nixos systemd[1]: iptsd.service: Main process exited, code=exited, status=254/n/a
Jun 29 22:41:08 nixos systemd[1]: iptsd.service: Failed with result 'exit-code'.
warning: error(s) occurred while switching to the new configuration
```

It seems that nixOS is trying to call the iptsd service but is unable to do so due to it not existing. I am unable to fix this issue.

fdgefgd, 2023-06-29:
I have also tried installing IPTSD through nixOS configuration.

```
environment.systemPackages = [
    pkgs.iptsd
  ];
```

It has no effect when I rebuild, but it seems to give the additional `iptsd` command, however I get the same error if I were to just run `iptsd` in the command line normally. However it does not seem to import the other 3 commands for some reason? `iptsd-find-service` `iptsd-find-hidraw`  `iptsd-check-device` do not become available after installing this package, but `iptsd` does? Not sure why this is considering they are all listed in the package in the repos.

Flameopathic, 2024-06-18:
is this still happening or can this issue be closed?

Hythera, 2026-11-08:
This issue is long resolved. No issues on SP7. Can be closed.