NixOS/nixos-hardware #672: Framework laptop battery drain

jackkav, 2023-07-17:
I have read that non-USB-C adapters can cause the drain, but also S0 sleep. I have attempted to use suspend-then-hibernate to some success but still seeing the battery taking a hit during sleep and potentially even shutdown. Wondeirng if anyone have gotten to the bottom of this and might form a useful contribution to this repo.

This is my current attempt.
```
services.logind = {
  lidSwitch = "suspend-then-hibernate";
  extraConfig = ''
    HandlePowerKey=suspend-then-hibernate
    IdleAction=suspend-then-hibernate
    IdleActionSec=2m
  '';
};
systemd.sleep.extraConfig = "HibernateDelaySec=2h";
```

Mic92, 2023-07-17:
I also have an issue to suspend on my Framework, 11 Intel Gen using anything else but the LID (i.e. external keyboard buttons or systemctl suspend). The suspend works but than I cannot wake it up again. I wondering if this also happens here with the `HandlePowerKey=suspend-then-hibernate` line.

AkechiShiro, 2023-09-09:
I'm not sure if this fix could help : 

Adding the [kernel parameter](https://wiki.archlinux.org/title/Kernel_parameter) acpi_osi="!Windows 2020" fixes a regression in s2idle, making the keyboard backlight and power button correctly turn off when suspending, as well as decreasing battery drain in suspend to about 1%/hour[[7]](https://community.frame.work/t/solved-laptop-stays-on-in-sleep/17168).

If your system fails to wake up from suspend due to the SSD disconnecting, you can try putting nvme.noacpi=1 in your [kernel parameters](https://wiki.archlinux.org/title/Kernel_parameters). This may also improve battery life in suspend, if your SSD has a bad ACPI implementation.

Sourced from the Archwiki.

Myned, 2024-02-26:
A [newer workaround](https://wiki.archlinux.org/title/Framework_Laptop_13#Suspend-then-hibernate_on_AMD_version) was added to the ArchWiki for the AMD version.

Add this to the kernel parameters until kernel 6.8 releases with [this commit](https://github.com/torvalds/linux/commit/3d762e21d56370a43478b55e604b4a83dd85aafc):
```
rtc_cmos.use_acpi_alarm=1
```

huantianad, 2024-03-22:
In the common intel config, it currently contains
```
# Fixes a regression in s2idle, making it more power efficient than deep sleep
"acpi_osi=\"!Windows 2020\""
```

But in the 12th gen intel config, we set `"mem_sleep_default=deep"`. 

These two settings seem to be in conflict with each other, so I'm not sure if `s2idle` actually gives better sleep battery life. Which one is correct?