NixOS/nixos-hardware #677: Orange Pi 5 plus support

adminy, 2023-07-26:
Hi, I hope this hardware is supported soon.

I have an [out of date repo](https://github.com/ryan4yin/nixos-rk3588) which supported rk3588 but this is for rk3588s. Not sure what the rockchip support is here, gpio should be similar to the Pi 4. the one that's very similar to it is `friendlyarm/nanopc-t4` not sure if it can be re-used somehow since they're both rockchip. Should there be maybe a more general name like rockchip if there is support?

Thank you.

IvanMalison, 2023-08-02:
That flake is definitely interesting, but it seems to be a strange mix of things that allow orangepi to work and things that the author, @ryan4yin personally uses. Here, for example, he's setting the default editor to neovim https://github.com/ryan4yin/nixos-rk3588/blob/0a2274989648cc5c925f074a295d0957e69e4545/orangepi5.nix#L241

I think a good first step might be to distill whatever is going on there into something that could be added to this repo.

IvanMalison, 2023-08-02:
Would be pretty cool of @ryan4yin could give some details of how well his setup is working.

adminy, 2023-08-02:
I have the orange pi 5 plus. I could try and run the code. also he's got a separate repository with newer changes. Hopefully its reproducible

ryan4yin, 2023-08-02:
Well, I tried to adapt the orangepi 5 plus about a few months ago. I mainly adapted based on the work of armbian, but when I encountered some problems at that time, I put aside this part of the work...

Some relevant information:

1. https://github.com/ryan4yin/nixos-rk3588/blob/dev/pkgs/kernel/README.md
2. https://github.com/fb87/nixos-orangepi-5x/tree/v5.10.x
3. https://nixos.wiki/wiki/NixOS_on_ARM/Orange_Pi_5

IvanMalison, 2023-08-02:
@ryan4yin so did everything work as expected with the normal pi 5?

ryan4yin, 2023-08-02:
@IvanMalison Only CPU confirmed. 
GPU was supported by[aciceri/rock5b-nixos](https://github.com/aciceri/rock5b-nixos), but I haven't tested it, I haven't tried NPU and GPIO either.