NixOS/nixos-hardware #678: linux-firmware guard fails probably due to new versioning

chaoflow, 2023-07-27:
https://github.com/NixOS/nixos-hardware/blob/ba9650b14e83b365fb9e731f7d7c803f22d2aecf/lenovo/thinkpad/p14s/amd/gen2/default.nix#L15C97-L15C111

Current linux-firmware version is "2023-07-24"

SuperSandro2000, 2023-07-27:
https://github.com/NixOS/nixpkgs/pull/245305

MangoIV, 2023-07-28:
is there a solution for this?

samueldr, 2023-07-28:
This would be intrinsically fixed the next time this package gets a version tag, but break again once it switches to another untagged release.

The conditional could check for `lib.hasPrefix "unstable" linux-firmware.version` and assume any untagged release is fresh enough.

Also, is this still required? Are there any supported NixOS releases matching `lib.versionOlder pkgs.linux-firmware.version "20230210"`? The simpler solution could be to just delete the check.

MangoIV, 2023-07-28:
The latter would also be my suggestion.

closed 2023-07-30 by Mic92