NixOS/nixos-hardware #684: VisionFive 2 visionfive2-firmware-update-flash spl.bin too big

skyrgobbler, 2023-08-02:
After getting my VisionFive 2 running NixOS and booted I wanted to update the flash (which currently only contains the JH7100 tool) to use it to boot from the eMMC, so I installed the firmware package as described at the end of the starfive/visionfive/v2 README, which compiled properly etc. Then when running `sudo visionfive2-firmware-update-flash` I get
```
/nix/store/yn59zvsrzlhd69fjg00bnc0gvx9dbv0b-starfive-visionfive2-spl-riscv64-unknown-linux-gnu/share/starfive-visionfive2/spl.bin won't fit into /dev/mtd0!
```
To check, the result of `ls -lr /nix/store/yn59zvsrzlhd69fjg00bnc0gvx9dbv0b-starfive-visionfive2-spl-riscv64-unknown-linux-gnu/share/starfive-visionfive2` is
```
.r--r--r-- 132k root  1 janv.  1970 spl.bin
```

I'm not quite sure where to go from here. Any help would be much appreciated (maybe @NickCao)

NickCao, 2023-08-03:
I see that the upstream provided spl file is of 129 KB (https://github.com/starfive-tech/VisionFive2/releases/tag/VF2_v3.1.5), so our spl is not that big. May you check the partitioning of your flash (https://doc-en.rvspace.org/VisionFive2/Quick_Start_Guide/VisionFive2_SDK_QSG/spl_new.html#updating_spl_and_u_boot-vf2__section_zpj_cqt_yvb) and see if the process described at https://doc-en.rvspace.org/VisionFive2/Quick_Start_Guide/VisionFive2_SDK_QSG/recovering_bootloader%20-%20vf2.html helps.

leifhelm, 2023-08-04:
This will be fixed with #646

closed 2023-08-06 by Mic92