NixOS/nixos-hardware #689: Asus ga401 broken

alexisquintero, 2023-08-05:
It seems to be broken since #580 was merged
Xserver fails to start, although journalctl is not exactly helpful, or more likely, I'm failing to make good use of it.

Mic92, 2023-08-06:
cc @LostAttractor

Mic92, 2023-08-06:
Does unsetting any of this help?

```nix
{
  hardware.nvidia = {
    # PCI-Express Runtime D3 Power Management is enabled by default on this laptop
    # But it can fix screen tearing & suspend/resume screen corruption in sync mode
    modesetting.enable = lib.mkDefault true;
    # Enable DRM kernel mode setting
    powerManagement.enable = lib.mkDefault true;

    prime = {
      amdgpuBusId = "PCI:4:0:0";
      nvidiaBusId = "PCI:1:0:0";
    };
  };
}
```

alexisquintero, 2023-08-06:
with either or both `modesetting` & `powerManagement` it still fails.
Trying to set the buses to `""` fails on the `switch` with
`error: cached failure of attribute 'nixosConfigurations.nixos-g14.config.system.build.toplevel'`

for completeness, the output of `lspci` (which seems to match with the values in the pr)
```
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Renoir IOMMU
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:01.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe GPP Bridge
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe GPP Bridge
00:02.4 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe GPP Bridge
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir Internal PCIe GPP Bridge to Bus
00:08.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir Internal PCIe GPP Bridge to Bus
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 51)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 7
01:00.0 VGA compatible controller: NVIDIA Corporation TU106M [GeForce RTX 2060 Max-Q] (rev a1)
01:00.1 Audio device: NVIDIA Corporation TU106 High Definition Audio Controller (rev a1)
01:00.2 USB controller: NVIDIA Corporation TU106 USB 3.1 Host Controller (rev a1)
01:00.3 Serial bus controller [0c80]: NVIDIA Corporation TU106 USB Type-C UCSI Controller (rev a1)
02:00.0 Network controller: Intel Corporation Wi-Fi 6 AX200 (rev 1a)
03:00.0 Non-Volatile memory controller: Intel Corporation SSD 660P Series (rev 03)
04:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c5)
04:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Device 1637
04:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
04:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir USB 3.1
04:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir USB 3.1
04:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] Raven/Raven2/FireFlight/Renoir Audio Processor (rev 01)
04:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) HD Audio Controller
05:00.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)
05:00.1 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)
```

I guess if no one else has any problem then the most likely explanation is that some other setting in my config is in conflict with these changes

Mic92, 2023-08-06:
To clarify the options above are set by the module and are the only one related to graphics as far I can see.

LostAttractor, 2023-08-26:
I noticed this problem before, I didn't realize it was caused by #580. I'm using Wayland so I didn't go any further
Kernel mode setting (KMS) & Direct Rendering Manager (DRM) are both needed for RuntimeD3 Power Management, which can significantly improved battery life.
However I do the same on ArchLinux (Enabling KMS & DRM) did not cause any problems. In theory this shouldn't cause any problems, I need to find out further.

frankitox, 2023-09-02:
I managed to narrow down the problem to importing the `common/gpu/amd/default.nix` file. Which I'm not sure what it does. Also if I add `services.xserver.videoDrivers = ["nvidia"];` to my config it fixes the problem. I suppose is overriding [this](https://github.com/NixOS/nixos-hardware/blob/fe0ea73/common/gpu/amd/default.nix#L20-L20) line.

LostAttractor, 2023-09-04:
> I managed to narrow down the problem to importing the `common/gpu/amd/default.nix` file. Which I'm not sure what it does. Also if I add `services.xserver.videoDrivers = ["nvidia"];` to my config it [...]

`services.xserver.videoDrivers = ["nvidia"];` is already in [there](https://github.com/NixOS/nixos-hardware/blob/master/common/gpu/nvidia/default.nix)
Is what you're doing covering `services.xserver.videoDrivers = lib.mkDefault [ "modesetting" ];` this line?
In fact, this is [the default setting of this option](https://github.com/NixOS/nixpkgs/blob/master/nixos/modules/services/x11/xserver.nix#L279). Maybe we should only add `amdgpu` (and nvidia) to this option? 
At least the [NixOS wiki](https://nixos.wiki/wiki/AMD_GPU) tells me this is what I should do, but I don't understand what that means

LostAttractor, 2023-11-10:
Because I haven't used xorg for a long time, I forgot to solve this issue, sorry here.
I have rechecked this issues, I found the problem is from this [commit](https://github.com/NixOS/nixos-hardware/commit/630a8e3e4eea61d35524699f2f59dbc64886357d) (from @K900 )
If change `services.xserver.videoDrivers` from `modesetting` to `amdgpu`, everything works fine.

K900, 2023-11-10:
That sounds like an Nvidia driver bug then. Can you post the actual X server logs. From a failed attempt?

LostAttractor, 2023-11-10:
Here is the full systemd log of booting the system, starting an xorg session with gdm/gnome45 and crashing.
[log.txt](https://github.com/NixOS/nixos-hardware/files/13322759/log.txt)
And here is the part where the crash occurs
```
.......
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) modeset(0): Output eDP-1 connected
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) modeset(0): Output HDMI-1 connected
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) modeset(0): Using spanning desktop for initial modes
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) modeset(0): Output eDP-1 using initial mode 2560x1440 +0+0
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) modeset(0): Output HDMI-1 using initial mode 3840x2160 +2560+0
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (==) modeset(0): Using gamma correction (1.0, 1.0, 1.0)
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (==) modeset(0): DPI set to (96, 96)
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) Loading sub module "fb"
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) LoadModule: "fb"
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) Module "fb" already built-in
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (==) modeset(G0): Depth 24, (==) framebuffer bpp 32
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (==) modeset(G0): RGB weight 888
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (==) modeset(G0): Default visual is TrueColor
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) Loading sub module "glamoregl"
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) LoadModule: "glamoregl"
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) Loading /nix/store/x37jm8n7rdv3s4pmllz2mhxx38xbhsf5-xorg-server-21.1.9/lib/xorg/modules/libglamoregl.so
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) Module glamoregl: vendor="X.Org Foundation"
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]:         compiled for 1.21.1.9, module version = 1.0.1
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]:         ABI class: X.Org ANSI C Emulation, version 0.4
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]:         ABI class: X.Org ANSI C Emulation, version 0.4
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) modeset(G0): glamor X acceleration enabled on NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) modeset(G0): glamor initialized
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) modeset(G0): Output DP-1-1 has no monitor section
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) modeset(G0): EDID for output DP-1-1
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (==) modeset(G0): Using gamma correction (1.0, 1.0, 1.0)
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (==) modeset(G0): DPI set to (96, 96)
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) Loading sub module "fb"
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) LoadModule: "fb"
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) Module "fb" already built-in
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) UnloadModule: "amdgpu"
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (II) Unloading amdgpu
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (==) modeset(0): Backing store enabled
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (==) modeset(0): Silken mouse enabled
[... cut ...]
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2874]: (EE) Server terminated with error (1). Closing log file.
Nov 11 03:50:19 CALaptopG14 /nix/store/ldzzm6bpicp6kqqlmrhlc6p652q6rrh4-gdm-45.0.1/libexec/gdm-x-session[2862]: Unable to run X server
Nov 11 03:50:19 CALaptopG14 gdm-password][2764]: pam_unix(gdm-password:session): session closed for user lostattractor
Nov 11 03:50:19 CALaptopG14 gdm[1991]: Gdm: GdmDisplay: Session never registered, failing
```

K900, 2023-11-11:
Looks like some Gentoo people tracked it down to the modesetting driver trying to probe the Nvidia card and failing: https://bugs.gentoo.org/910169

Can you try `videoDrivers = [ "nvidia" "modesetting" ]`, in that order?

LostAttractor, 2023-11-21:
I tried `services.xserver.videoDrivers = lib.mkForce [ "nvidia" "modesetting" ];`
But it has no effect

voiceless-zell, 2024-05-01:
I have also been having issues with this config. I am having different issues though. ```May 01 14:32:56 G14 /nix/store/6lpk2r07qrv8n0mvq1dfjgbds2bihjax-nvidia-x11-550.76-6.6.28-bin/bin/nvidia-powerd[60431]: nvidia-powerd version:1.0(build 1)
May 01 14:32:56 G14 /nix/store/6lpk2r07qrv8n0mvq1dfjgbds2bihjax-nvidia-x11-550.76-6.6.28-bin/bin/nvidia-powerd[60431]: SBIOS support not found for NVPCF GET_SUPPORTED function
May 01 14:32:56 G14 /nix/store/6lpk2r07qrv8n0mvq1dfjgbds2bihjax-nvidia-x11-550.76-6.6.28-bin/bin/nvidia-powerd[60431]: No matching GPU found
May 01 14:32:56 G14 /nix/store/6lpk2r07qrv8n0mvq1dfjgbds2bihjax-nvidia-x11-550.76-6.6.28-bin/bin/nvidia-powerd[60431]: Failed to initialize Dynamic Boost
May 01 14:32:56 G14 /nix/store/6lpk2r07qrv8n0mvq1dfjgbds2bihjax-nvidia-x11-550.76-6.6.28-bin/bin/nvidia-powerd[60431]: Failed to detach GPU id 256
May 01 14:32:56 G14 /nix/store/6lpk2r07qrv8n0mvq1dfjgbds2bihjax-nvidia-x11-550.76-6.6.28-bin/bin/nvidia-powerd[60431]: Failed to initialize Dynamic Boost
May 01 14:32:56 G14 /nix/store/6lpk2r07qrv8n0mvq1dfjgbds2bihjax-nvidia-x11-550.76-6.6.28-bin/bin/nvidia-powerd[60431]: Failed to detach GPU id 256
May 01 14:32:56 G14 systemd[1]: nvidia-powerd.service: Main process exited, code=exited, status=1/FAILURE
May 01 14:32:56 G14 systemd[1]: nvidia-powerd.service: Failed with result 'exit-code'.
May 01 14:32:56 G14 systemd[1]: Failed to start nvidia-powerd service
```

Not sure if it is at all related.

ludat, 2024-05-05:
I've hit this issue and this change fixed it

```
videoDrivers = [ "nvidia" ]
```

nixos-discourse, 2024-10-20:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/nvidia-powerd-service-fails-no-matter-what-i-try/54640/1