NixOS/nixos-hardware #702: Gigabyte B5XX motherboards fail to suspend with NVMe drives

lelgenio, 2023-08-20:
Previous discussion: https://discourse.nixos.org/t/where-to-report-solved-issue/31902

On various Gigabyte motherboards on the B5XX line, with NVMe drives connected, the system will immediately wake up from suspend.

Here's are some examples of people with this problem: 
1. https://bbs.archlinux.org/viewtopic.php?id=271357
2. https://www.reddit.com/r/gigabyte/comments/p5ewjn/b550i_pro_ax_f13_bios_sleep_issue_on_linux/
3. https://forums.linuxmint.com/viewtopic.php?t=346257
4. https://forums.linuxmint.com/viewtopic.php?p=2117343#p2117343
5. https://artemis.sh/2023/04/09/b550i-pro-ax-linux-suspend-sleep.html

Sometimes `echo GPP0 | sudo tee /proc/acpi/wakeup` is proposed as a solution, but that does not work for everyone, a more general solution is:

```nix
services.udev.extraRules = ''
  ACTION=="add", SUBSYSTEM=="pci", DRIVER=="pcieport", ATTR{power/wakeup}="disabled"
'';
```

This disable wakeups from all PCIe ports, which should not be a big deal, right. Maybe this will affect wake-on-lan for PCI network cards?

nixos-discourse, 2023-08-21:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/where-to-report-solved-issue/31902/4

Mic92, 2023-08-21:
Ok. We are currently not so great at supporting single hardware components. It's mostly laptops that we support as a whole. However we could also just publish a profile for a motherboard series similar to what we do for graphic cards... I would close this issue for now, since it doesn't have anything actionable beyond having the issue & solution documented.

closed 2023-08-21 by Mic92