NixOS/nixos-hardware #703: `raspberry-pi."4".audio.enable = true;` broken: DTOVERLAY[error]: can't find symbol 'audio'

Ramblurr, 2023-08-26:
With these options

```
hardware.raspberry-pi."4".apply-overlays-dtmerge.enable = true;
hardware.raspberry-pi."4".audio.enable = true;
hardware.deviceTree.filter = "bcm2711-rpi-4-*.dtb";
boot.kernelPackages = pkgs.linuxPackages_rpi4;
```

It fails with:

```

building Nix...
building the system configuration...
these 3 derivations will be built:
  /nix/store/nziz4xlb0jdf11zxrpxghcr092bp09yw-dtbs-filtered.drv
  /nix/store/ilrh3myjvzdrj1161yy42y0q00fmwryk-device-tree-overlays.drv
  /nix/store/fqy0aj92zra5s20kydph2kg0754p97q6-nixos-system-ovos-23.05.3023.74e5bdc5478e.drv
building '/nix/store/nziz4xlb0jdf11zxrpxghcr092bp09yw-dtbs-filtered.drv'...
./broadcom -> /nix/store/qwjd22886siziv4cfg9yqhpk58arajgh-dtbs-filtered/./broadcom
'./broadcom/bcm2711-rpi-4-b.dtb' -> '/nix/store/qwjd22886siziv4cfg9yqhpk58arajgh-dtbs-filtered/./broadcom/bcm2711-rpi-4-b.dtb'
building '/nix/store/ilrh3myjvzdrj1161yy42y0q00fmwryk-device-tree-overlays.drv'...
./broadcom -> /nix/store/qp09vsgjqmymb3rs8my6n17268j17phv-device-tree-overlays/./broadcom
'./broadcom/bcm2711-rpi-4-b.dtb' -> '/nix/store/qp09vsgjqmymb3rs8my6n17268j17phv-device-tree-overlays/./broadcom/bcm2711-rpi-4-b.dtb'
Applying overlay hifiberry-dacplusadc to bcm2711-rpi-4-b.dtb... ok
Applying overlay rpi4-cpu-revision to bcm2711-rpi-4-b.dtb... ok
Applying overlay audio-on-overlay to bcm2711-rpi-4-b.dtb... DTOVERLAY[error]: can't find symbol 'audio'
error: builder for '/nix/store/ilrh3myjvzdrj1161yy42y0q00fmwryk-device-tree-overlays.drv' failed with exit code 1
error: 1 dependencies of derivation '/nix/store/fqy0aj92zra5s20kydph2kg0754p97q6-nixos-system-ovos-23.05.3023.74e5bdc5478e.drv' failed to build
```

Version info:
```
$ sudo nix-channel --list
nixos https://nixos.org/channels/nixos-23.05
nixos-hardware https://github.com/NixOS/nixos-hardware/archive/master.tar.gz
```

jmoggr, 2023-09-04:
I am experiencing the same issue.

ri0t, 2023-10-05:
Yep, here too. I also don't get any SPI or i2c devices :crying_cat_face:

Mic92, 2023-12-13:
Does upstream have an updated version of audio-on-overlay?

ManOfDiamond, 2024-01-21:
> Knowing virtually nothing about device trees, it looks like the symbol's been removed:
> https://github.com/raspberrypi/linux/blob/342c7ee49e862edc30c893f141f55b9211b7a43b/arch/arm/boot/dts/bcm2711-rpi-cm4s.dts#L290
> 
> ```
> audio = <&chosen>,"bootargs{on='snd_bcm2835.enable_hdmi=1',off='snd_bcm2835.enable_hdmi=0'}";
> ```
> 
> But if I'm reading this right, then `boot.kernelParams = [ "snd_bcm2835.enable_hdmi=1" ];` should be all we need.
> 
> 

I can confirm that this works

mzabani, 2024-02-10:
This worked for me as well, with the caveat that you can't keep `hardware.raspberry-pi."4".audio.enable = true;` in the configuration (perhaps this is obvious to most, but wasn't to me).

madorian, 2024-05-22:
I've added that kernelParams, but I still get no audio from Kodi. Is that all you did?

nixos-discourse, 2024-06-09:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/why-do-i-get-conflicting-definiton-values-for-hardware-devicetree-filter-for-pi4/45214/5

RossComputerGuy, 2025-02-01:
This is broken for me as well.

DocBrown101, 2025-03-28:
Is there a workaround to make the HDMI sound output work again?

doronbehar, 2026-08-16:
A workaround might be composable with https://github.com/NixOS/nixos-hardware/pull/1947 & https://github.com/NixOS/nixos-hardware/pull/1894 .