NixOS/nixos-hardware #704: Framework 12th-gen Lingering Hardware Issues

ddanon, 2023-08-27:
# Framework 12th-gen Lingering Hardware Issues

- Right click is not working
- Middle click not working
- Fingerprint scanner not recognized by fprintd (details below)

## Hardware Specifics

- Make: Framework
- Model: 13
- Processor: i7-1260P

## Functionality List (modeled after the [Arch wiki](https://wiki.archlinux.org/title/Framework_Laptop_13))

- [x] GPU (internal)
- [x] Wireless
- [x] Audio
- [ ] Touchpad
- [x] Webcam
- [x] Bluetooth
- [ ] Accelerometer (untested)
- [ ] Fingerprint reader
- [ ] Ambient light sensor (untested)

## Steps to recreate

1. Using channels add and update nixos-hardware channel:

``` bash
sudo nix-channel --add https://github.com/NixOS/nixos-hardware/archive/master.tar.gz nixos-hardware
sudo nix-channel --update
```

2. Then import an appropriate profile path in /etc/nixos/configuration.nix

```nix
imports = [
  <nixos-hardware/framework/12th-gen-intel>
  ./hardware-configuration.nix
];
```
3. Update

```
sudo nixos-rebuild switch
```

4. Reboot (for good measure)
5. Note issues
  - Touchpad partially functional (right/middle click not working)
  - `sudo fprintd-enroll` yields `Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice: No devices available`

ddanon, 2023-08-27:
Full config files:

```nix
# Edit this configuration file to define what should be installed on
# your system.  Help is available in the configuration.nix(5) man page
# and in the NixOS manual (accessible by running ‘nixos-help’).

{ config, pkgs, ... }:

{
  imports =
    [ # Include the results of the hardware scan.
      # ./framework.nix
      <nixos-hardware/framework/12th-gen-intel>
      ./hardware-configuration.nix
      ./nvidia.nix
      <home-manager/nixos>
    ];

  # Bootloader.
  boot.loader.systemd-boot.enable = true;
  boot.loader.efi.canTouchEfiVariables = true;

  networking.hostName = "nixos"; # Define your hostname.
  # networking.wireless.enable = true;  # Enables wireless support via wpa_supplicant.

  # Configure network proxy if necessary
  # networking.proxy.default = "http://user:password@proxy:port/";
  # networking.proxy.noProxy = "127.0.0.1,localhost,internal.domain";

  # Enable networking
  networking.networkmanager.enable = true;

  # Set your time zone.
  time.timeZone = "America/Indiana/Indianapolis";

  # Select internationalisation properties.
  i18n.defaultLocale = "en_US.UTF-8";

  i18n.extraLocaleSettings = {
    LC_ADDRESS = "en_US.UTF-8";
    LC_IDENTIFICATION = "en_US.UTF-8";
    LC_MEASUREMENT = "en_US.UTF-8";
    LC_MONETARY = "en_US.UTF-8";
    LC_NAME = "en_US.UTF-8";
    LC_NUMERIC = "en_US.UTF-8";
    LC_PAPER = "en_US.UTF-8";
    LC_TELEPHONE = "en_US.UTF-8";
    LC_TIME = "en_US.UTF-8";
  };

  # Enable the X11 windowing system.
  services.xserver.enable = true;

  # Enable the GNOME Desktop Environment.
  services.xserver.displayManager.gdm.enable = true;
  services.xserver.desktopManager.gnome.enable = true;

  # Configure keymap in X11
  services.xserver = {
    layout = "us";
    xkbVariant = "";
  };

  # Enable CUPS to print documents.
  services.printing.enable = true;

  # Firmware update service
  services.fwupd.enable = true;

  # Enable sound with pipewire.
  sound.enable = true;
  hardware.pulseaudio.enable = false;
  security.rtkit.enable = true;
  services.pipewire = {
    enable = true;
    alsa.enable = true;
    alsa.support32Bit = true;
    pulse.enable = true;
    # If you want to use JACK applications, uncomment this
    #jack.enable = true;

    # use the example session manager (no others are packaged yet so this is enabled by default,
    # no need to redefine it in your config for now)
    #media-session.enable = true;
  };

  # Enable touchpad support (enabled default in most desktopManager).
  services.xserver.libinput.enable = true;
  
  # Enable Flatpak globally
  services.flatpak.enable = true;

  # Needed for fingerprint login per https://discourse.nixos.org/t/how-to-use-fingerprint-unlocking-how-to-set-up-fprintd-english/21901
  services.fprintd.enable = true;
  services.fprintd.tod.enable = true;
  services.fprintd.tod.driver = pkgs.libfprint-2-tod1-vfs0090; 
  #services.fprintd.tod.driver = pkgs.libfprint-2-tod1-goodix; 

  services.hardware.bolt.enable = true;

  # Define a user account. Don't forget to set a password with ‘passwd’.
  users.users.ddanon = {
    isNormalUser = true;
    description = "ddanon";
    extraGroups = [ 
      "docker"
      "networkmanager" 
      "wheel" 
      "wireshark"
    ];
    packages = with pkgs; [
      docker
      firefox
      flatpak
      fprintd
      gnome.gnome-software
      git
      grub2_full
      htop
      mullvad-vpn
      nextcloud-client
      podman
      wireshark
      thunderbird
    ];
  };

  # Enable automatic login for the user.
  services.xserver.displayManager.autoLogin.enable = true;
  services.xserver.displayManager.autoLogin.user = "ddanon";

  # Workaround for GNOME autologin: https://github.com/NixOS/nixpkgs/issues/103746#issuecomment-945091229
  systemd.services."getty@tty1".enable = false;
  systemd.services."autovt@tty1".enable = false;

  # Allow unfree packages
  nixpkgs.config.allowUnfree = true;

  # List packages installed in system profile. To search, run:
  # $ nix search wget
  environment.systemPackages = with pkgs; [
    busybox
    iputils
    libfprint
    pciutils
    toybox
    usbutils
    wget
  ];

  # Some programs need SUID wrappers, can be configured further or are
  # started in user sessions.
  programs.steam = {
    enable = true;
    remotePlay.openFirewall = true; # Open ports in the firewall for Steam Remote Play
    dedicatedServer.openFirewall = true; # Open ports in the firewall for Source Dedicated Server
  };
  # programs.mtr.enable = true;
  # programs.gnupg.agent = {
  #   enable = true;
  #   enableSSHSupport = true;
  # };

  # List services that you want to enable:

  # Enable the OpenSSH daemon.
  # services.openssh.enable = true;

  # Open ports in the firewall.
  # networking.firewall.allowedTCPPorts = [ ... ];
  # networking.firewall.allowedUDPPorts = [ ... ];
  # Or disable the firewall altogether.
  # networking.firewall.enable = false;

  # This value determines the NixOS release from which the default
  # settings for stateful data, like file locations and database versions
  # on your system were taken. It‘s perfectly fine and recommended to leave
  # this value at the release version of the first install of this system.
  # Before changing this value read the documentation for this option
  # (e.g. man configuration.nix or on https://nixos.org/nixos/options.html).
  system.stateVersion = "23.05"; # Did you read the comment?

}
```

ddanon, 2023-08-27:
Solving my own problem with the fingerprint scanner -- newbie mistake. I duplicated the library hardware configuration with my own configuration below. 

### /etc/nixos/configuration.nix
Changing from 
```nix
  # Needed for fingerprint login per https://discourse.nixos.org/t/how-to-use-fingerprint-unlocking-how-to-set-up-fprintd-english/21901
  services.fprintd.enable = true;
  services.fprintd.tod.enable = true;
  services.fprintd.tod.driver = pkgs.libfprint-2-tod1-vfs0090; 
  #services.fprintd.tod.driver = pkgs.libfprint-2-tod1-goodix; 
```

To
```nix
  # Needed for fingerprint login per https://discourse.nixos.org/t/how-to-use-fingerprint-unlocking-how-to-set-up-fprintd-english/21901
  #services.fprintd.enable = true;
  #services.fprintd.tod.enable = true;
  #services.fprintd.tod.driver = pkgs.libfprint-2-tod1-vfs0090; 
  #services.fprintd.tod.driver = pkgs.libfprint-2-tod1-goodix; 
```
fixed my problem and I was able to enroll fingerprint login in GNOME.

Still trying to figure out the issue with the touchpad.

mkg20001, 2023-08-30:
I have a 13th generation framework and the issue occurs here aswell

AkechiShiro, 2023-09-09:
Is there anyone that has reported this upstream ?

Nyabinary, 2023-09-11:
Hmm wdym to GNOME? @AkechiShiro

AkechiShiro, 2023-09-11:
@Nyabinary this is a gnome issue, it says hardware issue on the title, not sure what is about GNOME here.

ddanon, 2023-09-12:
@Nyabinary my fingerprint scanner is actually working flawlessly now that I'm using the upstream config rather than defining my own `services.fprintd`additions. Can you post the related info in your config files?

IRT touchpad issues: again it was a me problem. I was able to fix the behavior in gnome-tweaks by setting "mouse click emulation" to "area".

[image]

Nyabinary, 2023-09-13:
Same here I'm using the default config not adding anything to fprintd, I mean it works, but it always causes the failure when I first boot the computer as shown. For some reason I have to sign in through fingerprint or else it won't let me through even when typing my password manually.

ddanon, 2023-09-24:
That is funky... have you tried clearing all stored fingerprints and adding a new fingerprint through the GNOME GUI?

Nyabinary, 2023-09-24:
yep did that

closed 2023-10-23 by ddanon

Nyabinary, 2023-10-23:
> In GNOME, I still need help with the fingerprint scanner acting funky. Every time I restart or boot my computer, fingerprint auth automatically fails for some reason. It then takes 2 seconds before I can actually log in with my fingerprint. [image]

I still face this issue not a deal breaker but *very* annoying.

carletes, 2023-10-27:
> In GNOME, I still need help with the fingerprint scanner acting funky.

I've always seen this in my 11th-gen Framework as well.

Nyabinary, 2023-10-27:
> > In GNOME, I still need help with the fingerprint scanner acting funky.
> 
> I've always seen this in my 11th-gen Framework as well.

So clearly all the problems haven't been resolved *at least not for everyone*

AkechiShiro, 2023-10-28:
Maybe a missing fwupd update for the firmware driver of the fingerprint scanner @carletes, have you checked that ?
There seems to be also issues if the fingerprint device had a fingerprint enrolled previously on another distro and that it wasn't deleted in a specific way, I believe (I don't have any Framework laptop however, was just browsing through their blog)

carletes, 2023-10-28:
> Maybe a missing fwupd update for the firmware driver of the fingerprint scanner @carletes, have you checked that ?

My firmware's up-to-date, and after clearing my previously-enrolled fingers and enrolling again, I see the same behaviour as in https://github.com/NixOS/nixos-hardware/issues/704#issuecomment-1699111268: I's only after a couple of seconds that I'm able to use my fingerprint to log in (so it's not the end of the world).

This only happens with GDM (the Gnome Display Manager, which is what I'm using in my laptop). At no other time have I had issues with fingerprint recognition,  so perhaps it's more of an issue with GDM itself.

(and thanks a lot for your help anyway, @AkechiShiro, much appreciated!)

Nyabinary, 2023-10-28:
> > Maybe a missing fwupd update for the firmware driver of the fingerprint scanner @carletes, have you checked that ?
> 
> My firmware's up-to-date, and after clearing my previously-enrolled fingers and enrolling again, I see the same behaviour as in [#704 (comment)](https://github.com/NixOS/nixos-hardware/issues/704#issuecomment-1699111268): I's only after a couple of seconds that I'm able to use my fingerprint to log in (so it's not the end of the world).
> 
> This only happens with GDM (the Gnome Display Manager, which is what I'm using in my laptop). At no other time have I had issues with fingerprint recognition, so perhaps it's more of an issue with GDM itself.
> 
> (and thanks a lot for your help anyway, @AkechiShiro, much appreciated!)

Same here firmware is up-to-date clearing previously enrolled fingers and same problem.

AkechiShiro, 2023-10-28:
Thanks for having done all those checks as well @carletes @Nyabinary, this must probably be an issue with GDM then, maybe changing GDM to something else and trying to reproduce the issue, would allow us to narrow it down or not to GDM only.

But, I'm not sure if you're willing to debug why there is a small delay (it's not like a big issue, minor issue), maybe an issue should be opened upstream ? To get help from GDM folks in order to investigate ?

nyabinary, 2023-12-30:
> Thanks for having done all those checks as well @carletes @nyabinary, this must probably be an issue with GDM then, maybe changing GDM to something else and trying to reproduce the issue, would [...]

The issue also prevents me from typing in my password manually too, like it rejects my password (even though it corrects), forcing me to use the fingerprint scanner.