NixOS/nixos-hardware #713: Asus ROG Strix GL702VSK - built-in keyboard not working during LUKS decryption

Zytekaron, 2023-09-04:
OS: NixOS 23.05.3242.da5adce0ffaf (Stoat) x86_64
Host: ASUSTeK COMPUTER INC. GL702VSK
Kernel: 6.1.47

I used the Calamares installer from the live CD to set up nix on a flash drive with standard partitioning and encryption enabled. Upon booting, it reaches the LUKS decryption step and prompts for a password, but the laptop's built-in keyboard isn't able to type into it. I connected an external keyboard, and everything seems to be working fine beyond that point.

I tried including `asus/rog-strix/g733qs`, just in case the different laptop model had similar issues, but it didn't and just caused more issues with the installation, so I'm back to square one.

I'm happy to provide any additional information that would help debug or fix this. I may be able to assist with testing, but I'm not thoroughly familiar with NixOS yet.

Mic92, 2023-09-27:
Probably this response reaches you too late, but you can run `lsmod` on a booted installer (with working keyboard). Then add `boot.debug1devices` to the kernel parameters of your installed system (in grub/systemd you can press a key to edit boot entries for one boot only). `boot.debug1devices` will allow drop you in a rescue shell in which you can also type lsmod and compare if there are drivers missing from the first prompt. This usually helps you to get devices working in initrd.

Mic92, 2023-09-27:
Also check `dmesg` in case there is missing firmware. But I have never seen this so far keyboards.

Mic92, 2023-09-27:
Hope this helps.

closed 2023-09-27 by Mic92