NixOS/nixos-hardware #716: Pinephone/Pinephone Pro Support

luigi311, 2023-09-09:
Just saw librem5 support added so figured i would mention bringing in support for the Pine64 mobile phone devices. There is currently support for the Pinephone and there is a PR for Pinephone Pro in the mobile-nixos repo but it might be better to bring in support to the official nixos hardware instead of just leaving it there.

One of the big benefits i see in their repo is they have built their own installer so its possible to build an image and install it on to the sd card and the installer will prompt for all the information and flash it onto the emmc.

samueldr, 2023-09-09:
Mobile NixOS is a bit more than that,

What Mobile NixOS brings to your everyday system is those three main things:

 - hardware-specific knowledge (which could interchangeably live anywhere)
 - bootloader/generations problems workarounds
 - touch-based stage-1 implementation

The second point includes, for the Pinephones (and would help with the Librem 5) a generation selection mechanism *for stage-2* that is usable without requiring access to serial.

Why would you want a touch-based stage-1 implementation? For LUKS passphrase input for FDE.

Though, I don't think anyone should see this a preventing them to try and do things in other ways. They are only one solution to different problems.

Though note that some parts can be used independently. For example, I use the touch-based stage-1 implementation on my Steam Deck so I can input the FDE passphrase on boot without requiring a physical keyboard.

luigi311, 2023-09-11:
Thats all good information, i love the installer you have created since it makes it easier to get started on a mobile device like the PP. All three of those benefits that mobile NixOS sound great and seems like a great reason for Mobile NixOS to exist and continue existing but im not sure if hardware configurations should exist there permanently atleast for the more linux focused devices such as the PP/PPP and/or librem devices. 

We can add the configurations on here and then point to Mobile NixOS for documentation/instructions for getting an image with the installer and with all the other benefits that Mobile NixOS provides that you mentioned.