NixOS/nixos-hardware #773: Reorganize framework folder (Breaking)

kjhoerr, 2023-11-01:
Some discussion of this started in #765 so I thought it might be best to break it out to a separate issue.

The default framework module (`framework`) is for the Framework 13 11th-gen Intel laptop. Since there are some hardware quirks unique to this laptop that aren't shared by the other gen Framework laptops, this is less than ideal.

Framework is also releasing a Framework 16 laptop which will share some of the same platforms with the 13 but different specs and features, so in anticipation of that it would be best to break them into separate categories. For the above PR we created a 13-inch folder under framework to add the new 7040-amd module, since the upcoming FW16 release would also be 7040-amd. That leaves the other Framework 13 generations under the framework folder and not organized in the 13-inch folder, presumably until that's fixed as a resolution to this issue. I would say there's nearly no chance there is a FW16 that shares an Intel gen platform with the FW13's we've had so far, but they're still in the wrong folder at the moment.

Maybe there's not much more discussion to be had of this, but based on the PR discussion I would hope we are able to reorganize the framework folder and serve warnings for users who are using the old modules. Alternatively, we create a new framework folder so module links can be kept intact and we use links as necessary to group the modules.

kjhoerr, 2023-11-01:
I think technically we could move the 12th gen and 13th gen modules into a 13-inch folder without introducing breaking changes since the modules from the flake would just need to be updated to point to the new location, but doing anything with the 11th gen module will probably require breaking something, since the flake module for 11th gen is just called `framework`.

There are some common configurations between all the gens (so far) that would be worth having as a shared module so it doesn't keep getting duplicated for each gen we add, and it would make sense to make that the default framework module, or some other common module off the base folder.

kjhoerr, 2023-11-10:
I'm preparing a PR for changes including an assertion module replacement for the current 11th gen Intel module inspired some by other breaking module changes I have found in this repo. Hope to put that up for review after the other PR is finished. In the meantime my changes are [here](https://github.com/kjhoerr/nixos-hardware/tree/fw-rework)

Mic92, 2023-11-11:
It's fine breaking configuration as long as we have clear error messages.

PR #776 (2023-11-14): Framework profiles rework
Closes #773 

###### Description of changes

- Add assertion for default export for `framework` to tell users to move to use `framework-11th-gen-intel`
- Add deprecation doc to describe what change is happening and why
- Move 12th, 13th gen Intel modules into 13-inch directory (the export modules from flake.nix stay the same, there's almost definitely not going to be other 12th/13th gen intel models available at this point)
- Refactor common configurations for all 13-inch or Intel or AMD into common modules

###### Things done

- [x] Tested the changes in your own Nix [... cut]
closed 2023-11-14 by Mic92