NixOS/nixos-hardware #785: Support for peripheral devices?

cbleslie, 2023-11-21:
:wave: Hello everyone!

I have one or two peripheral devices I would like to make modules for. The only problem is that it seems `nixos-hardware` seems to be about; "the whole computer." is there room in this repo for individual "brand" peripherals?

This snippet would be the first thing converted to a module. It's for Apples External Super-drive:
https://gitlab.com/-/snippets/3623394

Thanks for your time!

ulrikstrid, 2023-11-27:
Shouldn't this go into `nixpkgs` as a service instead?

cbleslie, 2023-12-04:
@ulrikstrid ,
This was my first thought. One could say the same thing for `nixos-hardware` repo as well right? Where is the line drawn? If the contribution is not welcome here, that's fine. I can focus on putting it in `nixpkgs`.

cbleslie, 2023-12-04:
> As I see it you want to add a general service that anyone (?) can use.
> This repo is a collection of known good configurations for different
> hardware.
>
> Not sure if that makes it clearer or not?

Much clearer. Thank you.

closed 2023-12-05 by cbleslie