NixOS/nixos-hardware #786: Support for Lenovo Legion 7 16achg6

bratfizyk, 2023-11-23:
I'm using Lenovo Legion 7 16achg6.
`nixos-hardware` repo already contains an entry for a very similar module 16ach6h [here](https://github.com/NixOS/nixos-hardware/blob/master/lenovo/legion/16ach6h/hybrid/default.nix)

I naively tried using it on my machine but after a reboot the X service failed to start.

After a quick investigation I realized that for 16ach6h AMD GPU is connected to PCI:6:0:0.

https://github.com/NixOS/nixos-hardware/blob/c3abafb01cd7045dba522af29b625bd1e170c2fb/lenovo/legion/16ach6h/hybrid/default.nix#L22

However, on my machine it's PCI:5:0:0.
Here's a link to my dot file:

https://github.com/bratfizyk/dotFiles/blob/master/legion/configuration.nix

If I'd like to create a PR adding support for my machine, shall I copy all the files required for 16ach6h and change the PCI for AMD GPU, or shall I link the files for the existing configuration and only override the filed that differ?

4JX, 2023-11-24:
Should be pretty similar looking config wise, though you probably don't want the `16ach6h`'s edid part, as that's a workaround to get higher refresh rates working. And there's also some other bits that might not affect your device, such as https://github.com/NixOS/nixos-hardware/pull/550/files

bratfizyk, 2023-11-25:
OK, thanks. I'll experiment with it and will let you know.
I guess I can create a branch in this repo and then refer to it using e.g. a flake like this, right?

```
inputs.nixos-hardware.url = "github:NixOS/nixos-hardware/myNewBranch";
```

4JX, 2023-11-25:
I don't think you'd have repo perms to do so, would be better to just point to your own fork in the meantime. I'd also advise to first try with the simplest of modules, ie just "common" `imports` and `hardware.nvidia.prime` before diving into having separate dedicated and hybrid setups.

bratfizyk, 2023-11-25:
Thanks, doing this now.

bratfizyk, 2023-11-25:
Ok, here it is:
https://github.com/bratfizyk/nixos-hardware/blob/master/lenovo/legion/16achg6/hybrid/default.nix

I copied `hybrid` config from the existing model and it almost worked. I had to comment out `nvidia-prime` common module, as I was unable to start X with it. Do you have an idea how can I check the exact error message my machien complained at?

Also, I added a few other bits and pieces that I had found somewhere else. Not sure it these are needed, so I also left them commented out.

Last thing: as I'm not too much used to this sort of low level stuff, can you recommend a tool which I could keep an eye on to see if changes to my `nix` config impacts machine's performance?

4JX, 2023-11-26:
> as I was unable to start X with it

Probably related to `amdgpu.loadInInitrd = lib.mkDefault false;` which is part of making the EDID thing work. Need to try submit a PR to fix that at some point. Do take into account that not including the prime thing means you're running off the nvidia GPU which will greatly affect battery life, try adding it again while leaving the "loadInInitrd" part commented out. Also check X's error log and search online as needed. 

https://github.com/4JX/nixos-config/blob/6f02677b27063d66a66d65a57ac2447a7b3e9a95/hosts/terra/xorg.nix#L25-L26

> Also, I added a few other bits and pieces that I had found somewhere else. Not sure it these are needed, so I also left them commented out.

`open` default to `false` already which is probably how you want to have it. The other two options are probably already set elsewhere, but can't hurt to have them explicitly declared.

> Do you have an idea how can I check the exact error message my machien complained at?

Check `journalctl` (ie. `journalctl -b 0 -r`) and look for X's error log from a TTY after a failed start.

 > as I'm not too much used to this sort of low level stuff, can you recommend a tool which I could keep an eye on to see if changes to my nix config impacts machine's performance?

No idea about those

bratfizyk, 2023-11-26:
I've found the error message. It says:

`attempting starting display server on vt 7 failed`

and then it refuses to start sddm.

Looking for a solution on the web now.

bratfizyk, 2023-11-26:
OK, I made it work. However, I was a bit experimenting without understanding too much, so whether or not I found a good solution, it's a different story :).

Here's the current state of my nixos-hardware fork: https://github.com/bratfizyk/nixos-hardware/blob/master/lenovo/legion/16achg6/hybrid/default.nix

In order to make it work I had to add the following lines to my configuration.nix:

https://github.com/bratfizyk/dotFiles/blob/2c2bc2d9f6f742565bad59731b6075418e88a44b/legion/configuration.nix#L41

However, I'm a bit confused, as in nixos-hardware in nvidia-prime/disable.nix there is:

```
boot.blacklistedKernelModules = [ "nouveau" "nvidia" "nvidia_drm" "nvidia_modeset" ];
```

So I have no idea and I'm confused :).
If I understand the idea of prime correctly, all the processes should be run on my AMD GPU, unless explicitly run on NVidia, right? If so, how can I check which GPU is now used by deafault in my system?

4JX, 2023-11-26:
> So I have no idea and I'm confused :).

You're importing `  ../../../../common/gpu/nvidia/prime.nix`, which doesn't import the `disable.nix` file with it, it is only when importing a folder path that you implicitly import `/path/to/folder/default.nix` which is not the case here.

You do however probably want to blacklist `noveau` in your config to use the nvidia driver though if you are not doing that already.

> amdgpu.loadInInitrd = lib.mkDefault false;

Again, remove this, you do not need it to get a working config. This is only for the device specific EDID override.

>   opengl = {
>    enable = true;
>    driSupport = true;
>    driSupport32Bit = true;
>  };

These are already set in https://github.com/NixOS/nixos-hardware/blob/b006ec52fce23b1d57f6ab4a42d7400732e9a0a2/common/gpu/amd/default.nix#L22-L24

> If I understand the idea of prime correctly, all the processes should be run on my AMD GPU, unless explicitly run on NVidia, right? If so, how can I check which GPU is now used by deafault in my system?

Yes, you can read more about it in the [nvidia docs](https://download.nvidia.com/XFree86/Linux-x86_64/435.17/README/primerenderoffload.html). The command you ran should be enough to determine the current "main" GPU.

> In order to make it work I had to add the following line to my configuration.nix:

You're probably running into the issue that `nixpkgs` and `nixos-hardware` likes to use `modesetting` [by default](https://github.com/bratfizyk/nixos-hardware/blob/325f68566ee87276dcd7cfe874826959cde5b81b/common/gpu/amd/default.nix#L20) for AMD GPUs, seeing as though you have the same incompatibility as me you can refer to https://github.com/NixOS/nixos-hardware/issues/628 and:
https://github.com/4JX/nixos-config/blob/6f02677b27063d66a66d65a57ac2447a7b3e9a95/hosts/terra/xorg.nix#L28-L32

You can probably directly include that line in the hardware config.

bratfizyk, 2023-11-26:
To add a few interesting parts:
* I currently use Plasma with Wayland. In this setup everything works as expected (though - as I already mentioned - I might be missing something under the hood)
* I tried switching to X11 and then only my built-in laptop screen works. The external display is not even discovered (I can't see it in Plasma Display Settings).

It's getting more and more interesting :)

4JX, 2023-11-26:
Nvidia prime offload won't play well with external displays, as (assuming it is similarly configured to my model) all the ports are wired to the nvidia GPU.

If you want to use external displays with your laptop you would need to use `prime.sync`/`prime.reverseSync`

bratfizyk, 2023-11-27:
> You're importing ` ../../../../common/gpu/nvidia/prime.nix`, which doesn't import the `disable.nix` file with it, it is only when importing a folder path that you implicitly import `/path/to/folder/default.nix` which is not the case here.

OK, thanks. I misunderstood the meaning of the `imports = [ ./. ];` line, but it seems only to load `default.nix` from this folder, right?

> Again, remove this, you do not need it to get a working config. This is only for the device specific EDID override.
> These are already set in 

Now I understand, thanks.

> You can probably directly include that line in the hardware config.
Yes, that's what I found a similar line in your configuration and that's why I decided to give it a go.

> If you want to use external displays with your laptop you would need to use prime.sync/prime.reverseSync
I tried both but neither worked for me, though I read online that reverseSync is what I'm searching for.

However, after some more investigation, I found out this piece of config that did the job for X11:

```
  specialisation = {
    external-display.configuration = {
      system.nixos.tags = [ "external-display" ];
      hardware.nvidia.prime.offload.enable = lib.mkForce false;
      hardware.nvidia.powerManagement.enable = lib.mkForce false;
    };
  };
```

I guess that since this is a part of my personal setup, it should be put in my personal configuration, not in nixos-hardware.

bratfizyk, 2023-12-10:
> However, after some more investigation, I found out this piece of config that did the job for X11:
>
> ```
>   specialisation = {
>     external-display.configuration = {
>       system.nixos.tags = [ [...]

Ignore me, now I understand what does it mean, sorry :).

I think the first version is good to go. I've created a PR for it (https://github.com/NixOS/nixos-hardware/pull/796)
@4JX many thanks for all the help.

bratfizyk, 2023-12-19:
Implemented via https://github.com/NixOS/nixos-hardware/pull/796

closed 2023-12-19 by bratfizyk

JakobEdvardsson, 2024-01-13:
I'm sorry for waking up the Issue, I have the Legion 7 16achg6 as well and was not able to get it working, however when changing "amdgpuBusId = "PCI:6:0:0";" to "amdgpuBusId = "PCI:5:0:0";" it worked. Isn't it weird that the same hardware uses different busId or is this a common thing?

bratfizyk, 2024-01-13:
Hi there,
First thing - don't apologize as you're doing the right thing (i.e. introducing some new value to the discussion).

Regarding your question: `amdgpuBusId` for `16achg6` is already set to `PCI:5:0:0` (see [here](https://github.com/NixOS/nixos-hardware/blob/b34a6075e9e298c4124e35c3ccaf2210c1f3a43b/lenovo/legion/16achg6/hybrid/default.nix#L23)). Are you sure you imported the right config? I know it's set to `PCI:6:0:0` for `16ach6h` (see [here](https://github.com/NixOS/nixos-hardware/blob/b34a6075e9e298c4124e35c3ccaf2210c1f3a43b/lenovo/legion/16ach6h/hybrid/default.nix#L22))

bratfizyk, 2024-01-13:
Also, from my very recent experience: when I had my BIOS updated by Windows living side-by-side on my machine, it also changed BIOS NVidia settings to `NVidia only`, which in represented by "Discrete" in BIOS. NixOS `doesn't work` for me in this mode and I had to go to BIOS and switch it to hybrid mode (which - again - in BIOS is represented by `Dynamic`).

JakobEdvardsson, 2024-01-13:
I mixed up the numbers,  I changing from  "amdgpuBusId = "PCI:5:0:0";" to "amdgpuBusId = "PCI:6:0:0";".  I'm using hybrid in the bios. It's weird that we have different BusID. Here is what I did to make it work, ignore the commit message, I mixed up the numbers hehe. [Commit](https://github.com/NixOS/nixos-hardware/commit/5cb7e65b6c726d4453db47a4da72f27eca8df27a)

bratfizyk, 2024-01-13:
:) Magic.

To confirm this is the case, can you try running `sudo lshw -c display` ?
E.g. for me it returns:

```
*-display
       description: amdgpudrmfb
       physical id: 0
       bus info: pci@0000:01:00.0
       logical name: /dev/fb0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list rom fb
       configuration: depth=32 driver=nvidia latency=0 mode=2560x1600 visual=truecolor xres=2560 yres=1600
       resources: iomemory:fa0-f9f iomemory:fc0-fbf irq:109 memory:d0000000-d0ffffff memory:fa00000000-fbffffffff memory:fc00000000-fc01ffffff ioport:3000(size=128) memory:d1080000-d10fffff
  *-display
       product: amdgpudrmfb
       physical id: 0
       bus info: pci@0000:05:00.0
       logical name: /dev/fb0
       version: c5
       width: 64 bits
       clock: 33MHz
       capabilities: pm pciexpress msi msix bus_master cap_list fb
       configuration: depth=32 driver=amdgpu latency=0 resolution=2560,1600
       resources: iomemory:fc0-fbf iomemory:fc0-fbf irq:32 memory:fc10000000-fc1fffffff memory:fc20000000-fc201fffff ioport:1000(size=256) memory:d1500000-d157ffff
```

4JX, 2024-01-13:
@bratfizyk Do you perhaps have 2 drives installed in your machine? And @JakobEdvardsson do you only have one?

JakobEdvardsson, 2024-01-13:
Output from "sudo lshw -c display":
I do however have two nvme ssd in my system, @bratfizyk  and my guess is that you only have one? You might be on to something @4JX 

`*-display
       description: amdgpudrmfb
       physical id: 0
       bus info: pci@0000:01:00.0
       logical name: /dev/fb0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list rom fb
       configuration: depth=32 driver=nvidia latency=0 mode=2560x1600 visual=truecolor xres=2560 yres=1600
       resources: iomemory:f80-f7f iomemory:fc0-fbf irq:130 memory:d0000000-d0ffffff memory:f800000000-fbffffffff memory:fc00000000-fc01ffffff ioport:3000(size=128) memory:d1080000-d10fffff
  *-display
       product: amdgpudrmfb
       physical id: 0
       bus info: pci@0000:06:00.0
       logical name: /dev/fb0
       version: c5
       width: 64 bits
       clock: 33MHz
       capabilities: pm pciexpress msi msix bus_master cap_list fb
       configuration: depth=32 driver=amdgpu latency=0 resolution=2560,1600
       resources: iomemory:fc0-fbf iomemory:fc0-fbf irq:35 memory:fc10000000-fc1fffffff memory:fc20000000-fc201fffff ioport:1000(size=256) memory:d1500000-d157ffff`

4JX, 2024-01-13:
Whoops got those mixed up, reverse the number of drives for each person :sweat_smile: 

Similar thing happened when making the config for my device, I really need to get around to polishing it at some point...

JakobEdvardsson, 2024-01-14:
How would we implement this? Should we have different configs, one for having 2 ssd and one if only having one? Is nix able to detect the number of ssd installed? In that case it could be configured dynamically.

bratfizyk, 2024-01-23:
Hi, sorry for responding late, as life happened to me and I forgot about this thread.
I confirm I have only one NVMe drive in my laptop, which seems to confirm our hypothesis.

I'm not aware of any way of having Nix to detect the number of NVMe drives active, but I'm a nix beginner. I'll try digging into this a bit.
For now the only solution I see is adding another configuration which overrides `amdgpuBusId`.

PS: Lenovo, why are you doing this?