NixOS/nixos-hardware #800: Dell XPS 13 9310 fails to suspend

cloudlena, 2023-12-17:
When running `systemctl suspend` on my Dell XPS 13 9310, the suspend fails. I see the following lines in my journal:

```txt
Bluetooth: hci0: SSR or FW donwnload time out
hci_uart_qca serial0-0: PM: dpm_run_callback(): acpi_subsys_suspend+0x0/0x60 returns -110
hci_uart_qca serial0-0: PM: failed to suspend: error -110
PM: Some devices failed to suspend, or early wake event detected
```

When I disable Bluetooth in the BIOS, everything works as expected, but as long as Bluetooth is enabled, I cannot suspend.

Mic92, 2024-01-07:
Does it help to run `rmmod` the bluetooth drivers before?

JohnMolotov, 2024-09-20:
I would try using https://github.com/NixOS/nixos-hardware/blob/master/dell/xps/sleep-resume/bluetooth/default.nix  if you haven't already.

Also try `cat /sys/power/mem_sleep` and see if it's in the wrong sleep state.