NixOS/nixos-hardware #817: Fix for framework 13 AMD white screen / flickering issue on kernels >6.1

RichardJActon, 2023-12-31:
I was encountering an issue with my screen flickering white, especially after resume from sleep and did some searching and found a solution that I think it would be good to have added to the framework 13 AMD config here.

I'm still pretty new to nix lang and this is my first issue on this project so I'm not quite sure how best to implement this.

```
# if using this or any version >6.1
boot.kernelPackages = pkgs.linuxPackages_latest;
# then this is needed to stop the flickering issue
boot.kernelParams = ["amdgpu.sg_display=0"];
```

Sources:

https://blabli.blog/post/2023/03/14/nixos-amd-raphael-igpu-screen-issues/

https://discourse.nixos.org/t/white-flickering-desktop-environments-with-amd-raphael-ryzen-7950x/26286/3

Mic92, 2023-12-31:
I think the kernel params option would go here: https://github.com/NixOS/nixos-hardware/blob/master/framework/13-inch/7040-amd/default.nix
We already set linux kernel to latest to bump it to bigger than linux 6.1.

Luflosi, 2024-02-15:
I created #866 to fix this.

commit 106d3fec (2024-02-16): framework AMD 7040: work around white screen / flickering issue on newer kernels

Fixes #817.

Updating the kernel if it is too old is already done by `common/cpu/amd/raphael/igpu.nix`, so I removed the redundant code.
closed 2024-02-16 by mergify[bot]