NixOS/nixos-hardware #825: Surface Go 3 - unsupported kernel version - 6.1.68 required

roukydesbois, 2024-01-04:
Hello!

As I have been informed in [this issue](https://github.com/linux-surface/linux-surface/issues/1339), the reason why my rebuild of nixos doesn't work should be related to the version of the kernel that the patches are applied to.

I don't think I'm yet knowledgeable enough to suggest a pull request myself, so could you please:
- tell me if I'm right assuming that the current state of the surface-go hardware configuration is 6.1.62 as it looks like [here](https://github.com/NixOS/nixos-hardware/blob/8e34f33464d77bea2d5cf7dc1066647b1ad2b324/microsoft/surface/common/kernel/linux-6.1.x/default.nix#L11C22-L11C22)
- help me in modifying the hardware configuration so that the kernel that is used is 6.1.68 (it is the one supported by the linux-surface team)

Thanks!

Mic92, 2024-01-05:
You probably just need to change this string? https://github.com/NixOS/nixos-hardware/blob/7fd5585e07386e8cf1f335b3950e52dc1f7d4b5a/microsoft/surface/surface-go/default.nix#L20

Mic92, 2024-01-05:
Ah. Here we actually build this version: https://github.com/NixOS/nixos-hardware/blob/7fd5585e07386e8cf1f335b3950e52dc1f7d4b5a/microsoft/surface/common/kernel/linux-6.1.x/default.nix

Mic92, 2024-01-05:
maybe @stigtsp has time for a bump?

roukydesbois, 2024-01-05:
> You probably just need to change this string?
> 
> https://github.com/NixOS/nixos-hardware/blob/7fd5585e07386e8cf1f335b3950e52dc1f7d4b5a/microsoft/surface/surface-go/default.nix#L20

I had indeed tried, but the value needs to be from a list, and you answered too fast already 😄

Mic92, 2024-01-05:
In the best case you just change the hash to an empty string and bump the version:

https://github.com/NixOS/nixos-hardware/blob/7fd5585e07386e8cf1f335b3950e52dc1f7d4b5a/microsoft/surface/common/kernel/linux-6.1.x/default.nix#L24

Than nix will complain with the correct hash that you can put in this file.

stigtsp, 2024-01-05:
_

> maybe @stigtsp has time for a bump?

Sure, I'll bump the kernels later today

stigtsp, 2024-01-05:
@roukydesbois is it possible for you to use 6.6.8 instead btw? i've been wondering why 6.1 is specifically set for the surface-go variants.

roukydesbois, 2024-01-05:
I'm happy to use a more recent kernel - as an FYI and if I understand well, the linux-surface team only supports 6.6.6: https://github.com/linux-surface/kernel/tags?after=v4.14.334

stigtsp, 2024-01-05:
> I'm happy to use a more recent kernel - as an FYI and if I understand well, the linux-surface team only supports 6.6.6: https://github.com/linux-surface/kernel/tags?after=v4.14.334

I think this is just the kernel version the patches were rebased against last.

Can you test the following in your config? If it works well, I suggest we remove the 6.1.x kernel to reduce the work in keeping them updated.

```nix
microsoft-surface.kernelVersion = "6.6.8";
```

roukydesbois, 2024-01-05:
I had to set the value to 6.6.8 (because it must be from a list of values and 6.6 was not in it), but for now it hasn't crashed as fast as it did with the 6.1 version - I'll update you here once the build is done.

roukydesbois, 2024-01-05:
Hello !

After a few hours (and cleaning a few gigs of space), the build work well!

I'll close this, and I imagine you're good to go to bump the correct versioin @stigtsp !

Thanks to both of you for the swift help!

closed 2024-01-05 by roukydesbois

stigtsp, 2024-01-06:
> After a few hours (and cleaning a few gigs of space), the build work well!

Good news! I'm gonna propose that we remove 6.1 in that case, and will bump to 6.6.10 soonish.

stigtsp, 2024-01-07:
- https://github.com/NixOS/nixos-hardware/pull/831