NixOS/nixos-hardware #827: Framework 13 AMD crashed and then refused to boot after using framework-13-7040-amd

reinux, 2024-01-05:
Thankfully, I was able to boot back in using my previous profile and then just comment out the entry in my flake. Cool!

[image]

I think it has something to do with the fact that I'm using a Lexar SSD as opposed to the one provided by Framework, though I'm surprised it would take issue at all with storage.

Mic92, 2024-01-09:
Maybe use `nixos-generate-config` and see if there are block device driver missing for your SSD.

reinux, 2024-01-09:
Could you clarify what exactly I'm looking for here? Nothing looks out of place as far as I can tell.

```
{ config, lib, pkgs, modulesPath, ... }:

{
  imports =
    [ (modulesPath + "/installer/scan/not-detected.nix")
    ];

  boot.initrd.availableKernelModules = [ "nvme" "xhci_pci" "thunderbolt" ];
  boot.initrd.kernelModules = [ ];
  boot.kernelModules = [ "kvm-amd" ];
  boot.extraModulePackages = [ ];

  swapDevices =
    [ { device = "/dev/disk/by-uuid/f43ba3d7-0265-48fb-abe4-6f2e2e941a0b"; }
    ];

  networking.useDHCP = lib.mkDefault true;
  # networking.interfaces.wlp1s0.useDHCP = lib.mkDefault true;

  nixpkgs.hostPlatform = lib.mkDefault "x86_64-linux";
  hardware.cpu.amd.updateMicrocode = lib.mkDefault config.hardware.enableRedistributableFirmware;
}
```

Mic92, 2024-01-10:
The most important line is:

`boot.initrd.availableKernelModules = [ "nvme" "xhci_pci" "thunderbolt" ];`

Otherwise your NVME will not be found.

reinux, 2024-01-10:
Thanks for the help so far.

Yep, that line is in there.

The whole file, sans long comments:

```
{ config, lib, pkgs, modulesPath, ... }:

{
  imports =
    [ (modulesPath + "/installer/scan/not-detected.nix")
    ];

  boot.initrd.availableKernelModules = [ "nvme" "xhci_pci" "thunderbolt" ];
  boot.initrd.kernelModules = [ ];
  boot.kernelModules = [ "kvm-amd" ];
  boot.extraModulePackages = [ ];

  fileSystems."/" =
    { device = "/dev/disk/by-uuid/9c35facf-87f0-45b2-a1c5-9897c3d7e9db";
      fsType = "ext4";
    };

  fileSystems."/boot" =
    { device = "/dev/disk/by-uuid/A3C3-6724";
      fsType = "vfat";
    };

  swapDevices =
    [ { device = "/dev/disk/by-uuid/f43ba3d7-0265-48fb-abe4-6f2e2e941a0b"; }
    ];

<interface>.useDHCP`.
  networking.useDHCP = lib.mkDefault true;
  # networking.interfaces.wlp1s0.useDHCP = lib.mkDefault true;

  nixpkgs.hostPlatform = lib.mkDefault "x86_64-linux";
  hardware.cpu.amd.updateMicrocode = lib.mkDefault config.hardware.enableRedistributableFirmware;
}
```

Mic92, 2024-01-11:
Ok. Than I don't know why including the hardware profile makes your nvme actively fail. Could be any of the lines in here: https://github.com/NixOS/nixos-hardware/blob/b34a6075e9e298c4124e35c3ccaf2210c1f3a43b/framework/13-inch/7040-amd/default.nix#L30

Maybe even the newer linux kernel?

reinux, 2024-01-13:
Well, for whatever reason, it seems to be happy with unstable channel but not 23.11. I haven't run into any other issues yet with unstable so I'll probably keep using this unless it asplodes.

Thanks again!

closed 2024-01-13 by Mic92