NixOS/nixos-hardware #829: Framework kernel module

gaykitty, 2024-01-06:
It might be useful to install the `framework-laptop-kmod` kernel module (which is currently in unstable) on framework laptops. The main feature that module adds is battery charge limit adjustment.

Mic92, 2024-01-06:
There should be probably an escape hatch to disable this optional functionality in case its build breaks for a new kernel version.

Mic92, 2024-01-06:
The kernel module also looks simple enough. I wonder if they have plans to just upstream it.

Mic92, 2024-01-06:
Also: Does framework-tool not expose the same functionality?

gaykitty, 2024-01-06:
Having it exposed through sysfs means tools like tlp know how to use it.

zyansheep, 2024-01-22:
I tried to figure it out myself:
```nix
{ stdenv, lib, fetchFromGitHub, kernel, kmod }:
stdenv.mkDerivation rec {
  name = "framework-laptop-kmod";

  src = fetchFromGitHub {
    owner = "DHowett";
    repo = "framework-laptop-kmod";
    rev = "a9e8db9";
    sha256 = "0ic5k1jqdsz6plya93ysax8hn54vky6d449gmrnzja0sz73cwbq2";
  };

  hardeningDisable = [ "pic" "format" ];                                             # 1
  nativeBuildInputs = kernel.moduleBuildDependencies;                       # 2

  makeFlags = [
    # "KERNELRELEASE=${kernel.modDirVersion}"                                 # 3
    "KDIR=${kernel.dev}/lib/modules/${kernel.modDirVersion}/build"    # 4
    "INSTALL_MOD_PATH=$(out)"                                               # 5
  ];

  meta = with lib; {
    description = " Kernel module to expose more Framework Laptop stuff ";
    homepage = "https://github.com/DHowett/framework-laptop-kmod";
    license = licenses.gpl2;
    maintainers = [ maintainers.zyansheep ];
    platforms = platforms.linux;
  };
}
```
but it doesn't seem load when I do this:
```nix
boot.extraModulePackages = with config.boot.kernelPackages; [
    (config.boot.kernelPackages.callPackage ./framework-laptop-kmod.nix {})
  ]; # Load battery limit kmod
  boot.kernelModules = [ "framework-laptop-kmod" ];
```
`modprobe: FATAL: Module framework-laptop-kmod not found in directory /run/booted-system/kernel-modules/lib/modules/6.6.10`

any ideas?

I'm not using default kernelPackages, I have zfs installed so I have:
`boot.kernelPackages = pkgs.zfs.latestCompatibleLinuxPackages;`

gaykitty, 2024-01-24:
framework-laptop-kmod is already in nixpkgs

Mic92, 2024-01-24:
I would merge a PR that includes the kernel module but with an option to disable it as needed.

gaykitty, 2024-01-25:
@Mic92 how should I expose such and option?